iNavFlight/inav #10331: VTOL Improvements

MiguelitoBeard, 2024-09-01:
## Current Behavior
VTOL is working as intended and a great help when flying in the mountains; auto launch as a plane and return to land in a small space as a tricopter. Landing with a wind is however a challenge, the craft does not point into the wind and this can make unstable behavior.

## Desired Behavior
A great improvement would be having the option for the craft to automatically turn into the wind for vertical landing. Using the already very useful wind direction.

## Suggested Solution
Add a CLI command that after RTH and Transition (That is already automated) forces the craft to rotate into the wind before vertical descent and landing.

## Who does this impact? Who is this for?
VTOL users that want to fly planes where there is no runway, but where there is a wind that makes VTOL less stable.

## Additional context
VTOL really is simple and impressive, I am impressed and many thanks, this will make it even more useful.

Jetrell, 2024-09-02:
>A great improvement would be having the option for the craft to automatically turn into the wind for vertical landing. Using the already very useful wind direction.

Multirotor navigation doesn't have the ability at present for wind estimation, as fixedwing does.
This is more challenging to implement for copters, using the present means to estimate wind speed and direction. Because copters can sit stationary and descend vertically... And the last good wind heading before the transition from fixedwing flight, may not be relevant when landing in copter mode.

MiguelitoBeard, 2024-09-02:
Your point is clearly valid, but decending where the front at least starts into the wind would be much better than  a random direction. Yes direction can change so not perfect, as you state.