iNavFlight/inav #10338: Inav target request for JHEMCU F405 NOXE V2

torju01, 2024-09-03:
Hello,
can we add inav support to this 20x20 F405 little board?
it have BMi Gyro
the betaflight target is  JHEF405PRO.hex
https://github.com/betaflight/unified-targets/blob/master/configs/default/JHEF-JHEF405PRO.config

Thank you very much for help

mmosca, 2024-09-03:
New targets need to follow the new hardware policy.

https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

maybe @jhemcu can chime in.

torju01, 2024-09-03:
wow i am very surprise.
usually in quicksilver and emuflight communities, targets are quite simple to achieve !

CesarDavmun, 2025-01-24:
Hi, do you know if the target is available for inav? I want to buy this fc

jhemcu, 2025-03-01:
Sure, write the JHEMCUF405 target in the INAV firmware list

torju01, 2025-03-07:
Hello,    the inav 8.0 JHEMCUF405 target is not ok for JHEMCU F405 NOXE V2 flight board:   inav do not show/use accelerometer, barometer and gps captors with it

sensei-hacker, 2025-03-07:
> Hello, the inav 8.0 JHEMCUF405 target is not ok for JHEMCU F405 NOXE V2 flight board: inav do not show/use magnetometer, barometer and gps captors with it

 Still https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

closed 2025-03-07 by sensei-hacker