iNavFlight/inav #10339: "Found no valid ESC configuration" BLHeliSuite32.

ADV-FPV, 2024-09-03:
Having recently finished my first build I am met by a puzzling issue: 

~ When powering the build, I have **always** been missing the final 2 motor start-up sequence beeps. ~

I **_am_** able to connect to the ESC on BLHeliSuite32 (with USB C and Battery connected), but when I do so, the program doesn't recognise any ESC's. BLHeli32 says that the ESC is connected and I am able to select the correct port (the same one that I use for BF configuration), however when attempting to "Read Setup" or "Check" the ESC, the read fails every time and I am left with an error code stating: "Found no valid ESC configuration" (See photo 1).

The only other error code I have ever received on BLHeliSuite32 says: 

"Serial Interface "m4wFClntf" connected successfully

Connection to ESC #1 Failed!
Please check data- und power-connections to the ESC!" (See photo 2).

I have already reached out to the Facebook FPV community but sadly am yet to find solution which has helped me progress any further on any of this. The troubleshooting I have carried out so far is detailed below:

- From minute 1 of build completion I have only ever received the FIRST 3 motor start up beeps. **ALWAYS** missing the final 2 (Throttle signal beeps). 
- The motors have **never** spun up. Whether plugging in with/without a smoke stopper, on BF motor reorder tab or performing motor tests on either BF or BLHeli32.
- The build has no shorts according to a smoke stopper and shows no continuity on a multimeter. All of the motor connections also show the same output reading on a multimeter.
- The FC Powers on via USB C and is recognised by Betaflight (Whilst connected to ESC via wire harness)
- The wire harness from the ESC - FC is confirmed correct wiring. 
- The FC has been flashed with the latest firmware and is on the correct motor protocol (DShot600)
- All of the settings changed in BF save onto the FC perfectly fine. 

Very, very much at a complete loss with it all at this point, so if anyone can see if I'm missing something painfully obvious, or anything a little less obvious for that matter, any advice/suggestions will be gladly received!

ESC: T-Motor F55A Pro iii 4in1 32BIT
FC: T-Motor F7 Pro

[image]
[image]
[image]

mmosca, 2024-09-03:
what does your mixer tab look like in INAV and are you using dshot?

ADV-FPV, 2024-09-03:
Good to know that I'm already being dim somewhere... I must admit; INAV is not a software which I had ever downloaded. Having done so, when I connect the FC via USB C I get the notification: "You need to upgrade your firmware before you can use the Setup Tab". Same notification for all other tabs... 

As far as I can tell the FC Firmware is on the latest Rev...

ADV-FPV, 2024-09-03:
@mmosca I have just flashed the latest INAV Firmware onto the FC. Below are screenshots of the MIXER and OUTPUTS tab.

[image]
[image]