iNavFlight/inav #10346: Add support for PrincipIoT F7 FC

lukas-principiot, 2024-09-08:
## Current Behavior
PrincipIoT F7 FC isn't supported
## Desired Behavior
FC is supported
## Suggested Solution
This FC is already supported in BF under target `PRINCIPIOTF7` [link](https://github.com/betaflight/config/pull/549)
We can provide samples and resources as required, I'll be emailing shortly.

## Who does this impact? Who is this for?
Users of the flight controller.
## Additional context
Product page: [https://principiot.com/flight-controller](https://principiot.com/flight-controller)

MrD-RC, 2024-09-09:
You need to let the manufacturer know that you would like an INAV target. Here is the new hardware policy. 

https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

lukas-principiot, 2024-09-09:
I am the manufacturer, I sent an email to the coredev account from my email lukas@principiot.com

closed 2024-10-30 by DzikuVx