iNavFlight/inav #10348: Configurator

jkenney66, 2024-09-10:
## Current Behavior
Receiver Tab is pulling the wrong information. When FC installed upside down and the mode is horizon, the servos appear to be reporting that the what the servos are receiving for a signal from the flight controller. Not what the flight controller is receiving from the transmitter. This had me confused as to why my servo centers were changing with reboots and couldn't center them.

## Steps to Reproduce
1. FC Installed improperly (upside down)
2. Horizon Mode

## Expected behavior
The receiver tab saying servo channels near 1500. 

## Suggested solution(s)
Pull the data from the transmitter side, not the receiver.

## Additional context

---

- FC Board name and vendor: Speedybee F405 wing mini
- INAV version string: 
- # version
# INAV/SPEEDYBEEF405WING 7.1.2 Jun 1 2024 / 00:47:23 (4e1e59eb)
# GCC-10.3.1 2

sensei-hacker, 2024-09-10:
This is a little unclear - and that lack of clarity may well be the problem.
In the title and the description, you specifically say Receiver tab. Which shows you what the receiver is getting from the transmitter. That's the INPUT to INAV.   Those values are before INAV does anything.

But then you're talking about servos. Servos are the opposite end of the chain. They are the OUTPUTS, after INAV is done running. There are no servos on the receiver tab.

The receiver tab should show pitch channel increasing when you push the pitch stick forward. It should show roll and yaw increase when you move those sticks to the right.  If that isn't the case, the channel is reversed in your transmitter and should be fixed in your transmitter.

It is absolutely normal that centered sticks report near 1500 - that's the value conventionally defined as center since around 1968 or so.

jkenney66, 2024-09-10:
I had the fc unknowingly upside down in the wing. Inav was in horizon mode. When I went to the receiver tab it looked like it was showing me the outputs. Not the inputs. Like it was trying to correct it's orientation. It took me a couple days to figure out what was happening. Once I corrected the orientation in inav everything went to 1500 like expected.Directions as in pitch and roll worked as expected. Thanks for looking at this.John 

Yahoo Mail: Search, Organize, Conquer

sensei-hacker, 2024-09-10:
I'm glad you have it working now!

It sounds like you may have ALSO reversed it in the radio. The receiver tab shows what's coming in from the radio. It doesn't know or care how, or if, you mounted a vehicle.

The Outputs tab shows the outputs.

closed 2024-09-14 by DzikuVx