iNavFlight/inav #10349: Matek 722 cli says shared gyros locked...Is there a command to unlock these?

hoppey, 2024-09-11:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [INAV Discord Server](https://discord.gg/peg2hhbYwN)
* [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
* [INAV Official on Telegram](https://t.me/INAVFlight)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

sensei-hacker, 2024-09-11:
I take it you are referring to a line in the output like this?

Gyros detected: gyro 1, gyro 2 locked dma shared

sensei-hacker, 2024-09-11:
If that's what you're referring to:

1) For Betaflight questions, you might get better info on a Betaflight forum, Discord, or their Github.

2) The Betaflight folks might tell you that "locked" means the loop of locked onto the rate of your gyro. Think "GPS lock" or "missile lock".  If you want to break that lock in order to have worse performance, I suppose you could damaging the interrupt pin on the gyo chip, but that seems like a very bad idea.

hoppey, 2024-09-11:
I just wondered if there is a command to unlock it

sensei-hacker, 2024-09-11:
> I just wondered if there is a command to unlock it

I don't know think so, because that would be pretty silly. But again since this is a Betaflight question, you could ask the Betaflight folks. You're on the INAV GitHub at the moment.

closed 2024-09-11 by sensei-hacker