iNavFlight/inav #10350: Servo outputs on SpeedyBee F7V3

Dron170493, 2024-09-11:
Hello!
How to use servo output on SpeedyBee F7V3 on INAV 7.1.2?

When i use timer outputs like this
[image]

My resource  is fine:
A15: MOTOR1 OUT
B00: FREE 
B01: FREE 
B02: SPI_PREINIT CS
B03: MOTOR2 OUT
B04: MOTOR3 OUT
B05: FREE 
B06: MOTOR4 OUT
B07: FREE 

But when I try to use servo output for M5-M8, it become something wrong.
I set timer3 for Servos, and Add 3 servos, like this:
[image]

And then i look at my resource:
A15: MOTOR1 OUT
B00: SERVO7 OUT
B01: FREE 
B02: SPI_PREINIT CS
B03: MOTOR2 OUT
B04: SERVO5 OUT- WRONG! It must bee Motor3 outputs!
B05: SERVO6 OUT
B06: MOTOR3 OUT - - WRONG! It must bee Motor4 outputs!
B07: MOTOR4 OUT 

**When i try to spin M3, in really spin M4. Because M4 is on B06 in really
When i try to spin M4, no motor not spin**

After set timer3 for servos, the B06 set MOTOR3, and B07 set MOTOR4 automaticly. Why it is happened?? 
How i used servo on FC?

mmosca, 2024-09-11:
Unfortunately, the hardware does not allow for easily mixing servos and motors for quads, if you want s1 to s4 to be motors 1 to 4.

Due to how the stm32 hardware works, outputs on the same timer need to be all motors or all servos.

You can either resolder your esc to follow what you see in the mixer tab, or get a different flight controller.

closed 2024-09-11 by mmosca

Dron170493, 2024-09-12:
Thacks, I solder my servo on PWM output on TBS crossfire nano, it's simple then resolder esc