iNavFlight/inav #10356: GPS not working in NEUTRONRCF435MINI

giou, 2024-09-13:
## Current Behavior

Just made the switch from Betaflight and the GPS is not working (was fully working in Betaflight) is wired to UART2, GPS is BK-880. The GPS icon turns red in Inav 7.1.2 with the hardware health check. Also tried latest dev as of right now Inav 8.0.0 Sep 12 2024 / 07:28:38 (f3da4872) and different baud rates with the same results.

dump: https://pastebin.com/yG3eZJkX

```
# version
# INAV/NEUTRONRCF435MINI 8.0.0 Sep 12 2024 / 07:28:38 (f3da4872) dev
# GCC-13.2.1 20231009
```

sensei-hacker, 2024-09-13:
Which UART do you have it on?

This sounds like #10333

giou, 2024-09-13:
> Which UART do you have it on?
> 
> This sounds like #10333 which was merged about a week ago. #10333 also references a few others.

UART2. Sounds very similar to  #9972

giou, 2024-09-17:
I changed the port to UART3 and kept INAV8 and now the GPS is working, making it obvious that INAV has an issue with UART2 (At least for GPS) in this FC.

bkleiner, 2024-09-26:
Damn, this seems to be the same issue as https://github.com/iNavFlight/inav/pull/10366 with the additional factor that the neutron targ uses two pins with _different_ alternative functions.
i will try to  come up with a pull request.

giou, 2025-03-13:
Back at it again. This time in weird ways.

I was flying and crashed I hear sensor lost sensor lost sensor lost whatever, walk to the quad and seems fine. I power cycle it and it reads GPS FAILURE in the OSD though maybe a cable went loose so tried to check in field but no luck. Went home, triple check the solder joints, re-solder everything and check continuity seems fine but still no luck. Tried changing back again from UART3 to UART2 like it was at the beginning (thought UART2 was fixed, maybe it is) still no luck. Tried gpspassthrough to connect with u-center and it connects. Tried updating to 8.0.1 rc1 still no luck. But the gps is not faulty and the connection is good because this time I flashed betaflight and it works.

giou, 2025-03-15:
Tried flashing back INAV and testing GPS for 2 days still doesn't work. How is this possible?? It was working previously in INAV and is working right now in Betaflight. Maybe it has something to do with POSHOLD? was using it the moment I crashed by pressing disarm on accident, while falling pressed arm again but not all the way to lock the switch, just enough to activate it (radiomaster pocket) POSHOLD alleviated the crash to the point of having no damage. Also tried restoring gps settings with u-center.

sensei-hacker, 2025-03-15:
Look on the GPS tab. Is the Messages count increasing?
Is the Errors count increasing?

The two most likely scenarios:

1. The crash disturbed the wire coming from the FC TX to the GPS rx. This would prevent INAV from applying appropriate settings, such as the best channel to use for each constellation. Betaflight may still receive whatever the GPS happens to send, because YOLO.

And/or:

2. While you had U-Center open you changed settings. Such that there is now an incompatibility with the optimum settings we've developed over the years.

A next step would be to reconnect U-Center and click the button in U-Center that resets the module to defaults, clearing any changes you made.

If U-Center is not able to communicate with the module, that wire is probably damaged. If it is able to successfully communicate, clicking that option will remove any incompatible settings you did.

giou, 2025-03-17:
> Look on the GPS tab. Is the Messages count increasing? Is the Errors count increasing?

No errors or messages count increasing.

> The two most likely scenarios:
> 
>     1. The crash disturbed the wire coming from the FC TX to the GPS rx. This would prevent INAV from applying appropriate settings, such as the best channel to use for each constellation. Betaflight may still receive whatever the GPS happens to send, because YOLO.

I highly doubt it because I re-solder everything 2 times, changed UART and checked continuity with multimeter for every pin.
About the RX only theory. I can use AssistNow from INAV, it reads:

> 2025-03-17 @ 14:24:41 -- 233/233 AssistNow messages sent.
> 2025-03-17 @ 14:24:41 -- AssistNow data transfer complete.

And with _gpspassthrough_ I can see the messages in the console about the menu I'm in.

```
And/or:

    2. While you had U-Center open you changed settings. Such that there is now an incompatibility with the optimum settings we've developed over the years.
```

The only setting I used in U-Center for these test was restoring to factory defaults.

Also tried INAV 7.1.2, same error.

sensei-hacker, 2025-03-17:
Can you show us your Ports tab, please, and the actual wiring?