iNavFlight/inav #10360: Zero distance after Gps lost and restore

Zimenkov, 2024-09-16:
While flying INAV 7.1 and Gps lost (sats=0) and back to normal (sats>0) distance is zero. Examples in video : https://youtu.be/6rszFYM9t9I 
There was no this problem in INAV6.1 Just appeared in INAV7.1 In video four examples (digital and analog). RTH after that doesn't work, no arrow to home. If activate RTH (with switch) it tries to land (landing mode). So I suppose its in INAV code not in hardware (as in 6.1 all was fine).
 How to reproduce: fly, lost gps. 
- FC Board name and vendor: matek wmn, speedybee f405 wing and other
- INAV version string: 7.1.1 7.1.2

MrD-RC, 2024-09-16:
It looks like it's resetting your home position. As the direction to home arrow changes to the "at home" indicator when the distance changes to 0. 

Check your `inav_reset_home` parameter. It should be `first_arm`. If that's ok, something else could be resetting the home position. 

https://github.com/iNavFlight/inav/blob/master/docs/Settings.md#inav_reset_home

Zimenkov, 2024-09-16:
Thanks. Yes, inav_reset_home = FIRST_ARM
Even if not - distance shouldn't be zero all the time. Even after I land and re-arm - it always zero. Just accumulator re-attach helps (reboot). It smth else and on different models.

breadoven, 2024-09-16:
It's more than just the home resetting, both vertical velocity and the altitude also freeze briefly before restarting.

Really needs a  log and Diff file to try and understand what's going on.

Zimenkov, 2024-09-25:
Here is from one flight:
 DVR video (gps lost sats for a second) from 0:14  https://youtu.be/Wi2jcM4uh9g

 LOG of this flight (in Excel) marked yellow line of sats lost sats) line 268 : https://docs.google.com/spreadsheets/d/1ybDZW68ncBZK_LETA_pu1NS1rVQmQWpD/

 DIFF of this copter :  https://drive.google.com/file/d/1Tzc4YjifGmcW6ytaDbLS8ok7G0zIuMKb/

it shows that sats came back and gps position in on osd, but distance to home is zero. And not changes while flying. Hope this will help. Thanks.

breadoven, 2024-09-26:
Needs an INAV blackbox log file. The log posted above just doesn't include the info that would be useful.

However, I've noticed you're using 7.1.1 and have `inav_use_gps_no_baro `set to the default of ON which causes problems, https://github.com/iNavFlight/inav/issues/10040. When the GPS drops out you end up with no valid altitude even with a working Baro. Easy solution is to set `inav_use_gps_no_baro `to OFF.

Can't see why this would affect the Home Distance though. The only thing that was noticeable in the video was Home Distance went to -0 briefly which is shouldn't do, should never be negative obviously. Smells a bit like an overflow problem somewhere.

Macianno, 2024-09-29:
I had similar situation in june with speedybee F7 v3 FC. I just described this here: https://github.com/iNavFlight/inav/issues/10391#issuecomment-2381576810 but additionally I have a problem with running away engines after RTH landing or trying return to home.

Macianno, 2024-10-01:
@breadoven here is a log with my strange gps reset: [LOG00030.TXT](https://github.com/user-attachments/files/17214109/LOG00030.TXT)

breadoven, 2024-10-01:
> @breadoven here is a log with my strange gps reset: [LOG00030.TXT](https://github.com/user-attachments/files/17214109/LOG00030.TXT)

OK it looks like there's an overflow on the Nav Position and Velocity estimations after the problem GPS loss. Pos x and y briefly overflow at signed 32 bit and the velocities at signed 16 bit (= 328 m/s). They then both go to 0 for the rest of the flight which is why it thinks it's always at Home and the Trip distance freezes (XY velocity used to calculate Trip Distance will also be 0). This didn't happen for the 2 GPS dropouts that happened before this, Pos and Velocity restarted normally once the GPS position was trusted again. Just need to work out why it's doing this now.

Macianno, 2024-10-01:
I just remind that I had few times also altitude reset when I change flying modes in PosHold... Maybe this is connected. I will try to record log for this. Maybe tomorrow if weather will be nicer ;)

breadoven, 2024-10-03:
I've had a look at this and I can't work out what the problem is. It might be to do with the fact it regained a valid GPS position briefly then lost it again briefly then recovered it which confused something in the code. However, the Blackbox logging rate in LOG00030.TXT wasn't 100% so something significant could have been missed, i.e. GPS glitch (Logged GPS data looks OK, no glitches).

Macianno, 2024-10-03:
OK, I will change blackbox logging to 100% and send it here if problem with gps reset occur again. I was flying 6 packs yesterday but altitude reset do not appear yet, so we have no more data for investigation ;)

b14ckyy, 2026-02-14:
Fixed in https://github.com/iNavFlight/inav/pull/11270

closed 2026-02-14 by b14ckyy