iNavFlight/inav #10361: VTOL Servo speed control

tohclin, 2024-09-16:
I uses a linear servo for my VTOL project, servo 4. Since the linear servo is slow, I need to control the servo speed.
The linear servo moves from profile2(2000us) to mixer transition mode(1500us), then to profile1(1000us) slowly as I wish.
But it moves from profile 1 to transition mode **instantly**, then to profile 2 slowly. I am not sure if this is a bug or not.

# mixer_profile 1
mixer_profile 1
# smix
smix 0 2 1 50 0 -1
smix 1 3 1 50 0 -1
smix 2 2 2 50 0 -1
smix 3 3 2 50 0 -1
smix 4 4 29 100 15 -1
smix 5 1 0 100 0 -1

# mixer_profile 2
mixer_profile 2
# smix
smix 0 2 1 50 0 -1
smix 1 3 1 50 0 -1
smix 2 2 2 50 0 -1
smix 3 3 2 50 0 -1
smix 4 4 29 -100 15 -1
smix 5 1 0 100 0 -1
smix 6 4 38 100 15 -1

SKelectronics, 2025-01-09:
same issue here, from vtol to transition to wing very good and smooth speed.
from wing to transition without speedcontrol, direct drive to transition position, further from transition to vtol speed as expected.

Pairan, 2025-01-13:
Yeah ... how do we get the other half of the Transmission mode back again? Where Do we need to dig to make that happen? Help and PRs appreciated :)

Like this  ...
[Video showing the described effect](https://youtu.be/ozcYxwLuXcE)

Pairan, 2025-01-23:
... anybody?

SKelectronics, 2025-01-23:
with 8.0 final the servos are moving a bit while transitioning, and same bug as above described.

from transition to FW = little bit backwards and then with slow speed to toward position
from FW to transition = max speed to transition, but a little bit too far, and then this little bit backwards to transition position

Pairan, 2025-03-11:
I was out testing this and flew the T1 with several times to check transitions. Currently it seems that in TRANSITION MODE from tri-copter to FW it is necessary to throttle down BEFORE going into final FW state! My T1 went like a Falcon 9 90° up after switching from TRANSITION to FW ... 

I would love to see that somebody would make this VTOL part in INAV to be more painless.

shota3527, 2025-03-13:
The servo speed will not work correctly when the mixer profiles are switched.
Servo speed requires storing an internal state to work; the mixer profile switch discards that internal state.
Carrying that internal state during the switching will introduce complexity to the source code. It was implemented in some test versions, but I removed it in the final release. If servo speed is needed in some configurations, i will consider to bring the functions back.
I recommend not using servo speed if possible for now
If the TRANSITION to FW causes a sudden pitch-up without servo speed, check that the trust center is close to the COG in the MC state. For the tri-rotor setup, the ideal COG will be 1/3 from the tilting rotor to the rear rotor. Also, consider reducing throttle mixing for the tilting rotor in the MC state(to shift the trust center backward)

Pairan, 2025-03-13:
great and valua reply @shota3527 - this helps to understand whats behind it! 

Are the GValues perhaps a thing to use for that internal state while transitioning? I know this may sound hilarious for now but perhaps this can take some of the complexity until a better solution is found.