iNavFlight/inav #10363: MSP Message for link stats of MSP-RC systems

b14ckyy, 2024-09-17:
## Current Behavior
Using a RC Link system like [mLRS ](https://github.com/olliw42/mLRS) that support RC over MSP, there is currently no way to get receiver and link information to the OSD and for blackbox logging. mLRS is the first system to provide this feature with ELRS maybe following in the future. 

## Desired Behavior
Having a MSP message the Rx can use to send RC Link stats and information to the FC. This would allow them to be logged and used for the OSD. 

## Suggested Solution
A New MSP message in INAV so it can receive RC Link information from MSP-RC receivers. 
The Link Statistics and Information should be send as a message string that can be extended in the future if necessary. One extra leading bit should be used, to allow the RC system to flag certain fields as valid or "not used", to allow the INAV side to be future proof. 
If more fields are added in the future, then older INAV versions should also work with the existing fields so a longer payload should just be cut off and additional fields ignored. 

Basic Fields to be available and their length (+1 leading bit for the ignore flag):
RSSI% - 7
RSSIdBm - 7
Rx LQ% - 7
Tx LQ% - 7
SNRdBm - 5 (-15 to +15 dBm, only shows when low or negative)
Rx Antenna - 1
Tx Antenna - 1
Band (2.4/900) - 1

Total Payload so far: 44bit

## Who does this impact? Who is this for?
All pilots who want to use MSP capable receivers. 

## Additional context
Feel free to post suggestions. We better have more fields supported than we currently need to not get into trouble in the future. 

@olliw42
@Scavanger

olliw42, 2024-09-17:
very desirable IMHO and suggestion looks highly reasonable to me 

👍

b14ckyy, 2024-09-17:
@OptimusTi thanks for the hint. you think any RF Chip can handle a Signal that is more than 15dB under the noise floor? I have seen mLRS in 31Hz mode to reach -8dB SNR without breakup and at -10 it was over. -15 will be a very long shot I guess. Even for LoRa. Sure we could add a bit more if that is really reasonable but I have doubts.

b14ckyy, 2024-09-17:
I think Crossfire was FSK only in the past and also had it but I could be mistaken. 
Anyway. If you think more range for SNR makes sense we can add a bit and get -32 to 32. We still have 1 bit left to have at least the ability to have 2 Link stat frames for dual operation, without raising the payload size. 

in the worst case if we do 4 Linkstat frames, we could steal a bit from the LQ. I think a 2% step precision for LQ would not hurt anyone :P Lets see what @MUSTARDTIGERFPV has in mind for the framing and what @Scavanger says about that implementation.

olliw42, 2024-09-17:
my mistake, I probably should have said specific to semtech sx chips, not ruling others do too but point is others don't do ... so, doesn't invalidate the statement ;)

OptimusTi, 2024-09-17:
How many sf used in mlrs? That should tell you how low you can go.

MUSTARDTIGERFPV, 2024-09-17:
For SNR, anything beyond -20dB is unlikely to be valuable. Crossfire measures SNR differently than ELRS and mLRS, not using RegPktSNR but instead measuring RSSI during the dead time between packets - that's why it has SNR available even in its 150Hz FSK mode.

On the MAVLink RC channels discussion, a device manufacturer was arguing for 4 sublink values, so we might want to keep that flexibility. IMHO it's more valuable to have this than SNRs that LoRa chips can't decode anyway (and are likely near theoretical limit).

b14ckyy, 2024-09-18:
@MUSTARDTIGERFPV I think you both have good points. In INAV's case and I would assume also in BF, the SNR on the OSD only shows up if it is <10dB. As we can only send integers anyway afaik, a 5 bit value for SNR should be fine and on the INAV side we shift that value by -20. This way we get a -20 to 11 range to use on the FC while the Rx sends a raw value from 0-31. 

Additionally a value of 31 could then be just "SNR not supported/disabled" as well.

b14ckyy, 2024-09-18:
So far this looks pretty good but I have one question: 

with the antenna Tx/Rx state, we would currently cover information about a dual-band or diversity setup with 2 transceivers. 
But if we now use the option to have sub-link IDs to get data for each receiver, it makes not much sense to have every tx/rx combination for 2 antennas. I wonder how we could optimize that, especially as we also have receivers that have one transceiver with antenna switching diversity. 

Would this make more sense then: 
`Receiver Antenna - 2 (Not Used; Rx; Tx; RxTx)`

So we would use each sub-link ID also for each antenna if there are multiple. Could even be used for antenna switching if you want that and we save 2 bit.

olliw42, 2024-09-18:
I think you encounter here a situation which comes from not being logically/conceptually consistent but mish mash. I.e., here mix the concept of sending data in different messages with the concept of sending some combined data in one message.

this may point to few obvious appraches to resolve :
- one could take the ID concept serious, and only send the data for that antenna in that message. One would then only have the info for that antenna in the message (i..e. one bit). A diversity or dual band (two antenna system in general) would have to send two messages.
- one could make the id field such as to allow indicating that the message may also reflect two antenna. E.g a bit mask of four bytes, there each bit is for one antenna. In one message only two bits could be set at most.
- one could reinterpret the ID to not mean one particular antenna but a cluster of antenna, like cluster 1 meaning antenna 1 and 2, cluster 2 meaning antenna 3 and 4.

not saying any is great, but just to indicate paths towards possible resolution

olliw42, 2024-09-18:
I think the vaue for power needs more specification. The power can also be negative, as well as positive. Not clear to me this is intended to be represented currently.

b14ckyy, 2024-09-18:
> * one could take the ID concept serious, and only send the data for that antenna in that message. One would then only have the info for that antenna in the message (i..e. one bit). A diversity or dual band (two antenna system in general) would have to send two messages.

That's actually how I had that in mind. From a users perspective, it does not really matter how many transceivers are on their hardware, all they see from the outside are antennas. Additionally I highly doubt that we will ever see a 4 transceiver system with antenna diversity on each (8 antennas total). In the end the idea is to provide 4 slots to write 4 different datasets and its up to the RC Link Code to decide how its used and its on the FC Software side to decide how the information are provided to the user. The main goal is to have a way to provide separate link stats for up to 4 transceivers. If you wanna use it for antennas in a Antenna switching diversity setup and one transceiver, that's also possible, but the only difference between the ID 0 and ID 1 would be, what antenna is active. All other values would be identical. 

> * one could make the id field such as to allow indicating that the message may also reflect two antenna. E.g a bit mask of four bytes, there each bit is for one antenna. In one message only two bits could be set at most.

Theoretically we could say we use the ID for transceiver info only and an additional Antenna bitmask as you said but then we would end up with a full extra byte in the message with not much benefit. In a 4 Transceiver setup that would be 4 bytes more for a full cycle. But that would only make sense for that unlikely setup of more than 4 total antennas. 

I hope I understood your point correctly. 

> * one could reinterpret the ID to not mean one particular antenna but a cluster of antenna, like cluster 1 meaning antenna 1 and 2, cluster 2 meaning antenna 3 and 4.

As above, that would be up to the RC link code. 

####################

My idea basically is something like this, in a multi-transceiver setup: 
[image]

To have one element for each transceiver that shows either RSSI or LQ as bars + Info if the transceiver is used for R/T/RT 
But this has to be defined later. For now we just talk about the way to feed INAV with the data. 

> I think the value for power needs more specification. The power can also be negative, as well as positive. Not clear to me this is intended to be represented currently.

You think Transmission power levels and negative dB are actually practical and will be used? For some "min" low power mode I would have said we just show 0dB. 
I mean sure if it makes sense we can shift the value by -20 like we did with SNR so you can reach -20dB transmit power if you think that makes sense. Gives us still ridiculous 43dB max.

olliw42, 2024-09-18:
power levels: wasn't attempting a suggestion just trying to say that it needs to be specified. :)

looking forward to 63 dBm devices LOL

b14ckyy, 2024-09-18:
got it, added that.

olliw42, 2024-10-09:
any progress planned, or dead horse?

b14ckyy, 2024-12-05:
Closing as implemented in #10451 for 8.0 RC1

closed 2024-12-05 by b14ckyy