iNavFlight/inav #10364: Dual gyros locked

hoppey, 2024-09-17:
I've a Matek 722se with 2 gyros ..INAV says no gyro,no accelerometer..Betaflight says gyros shared lock..GUI shows no gyro or accelerometer ..Tried everything to unlock them and still no gyro/accelerometer..Any ideas?

mmosca, 2024-09-17:
what gyro is reported by bf?

sensei-hacker, 2024-09-17:
> Tried everything to unlock them

Lock is a good thing. Think "GPS lock". You do not want to "unlock them".

You do want INAV to detect the gyro, of course. Just to confirm, which INAV target are you using, exactly?

hoppey, 2024-09-17:
BF doesn't tell me..It just says shared lock on gyros

hoppey, 2024-09-17:
INAV says none on gyro and accel..I Believe I have matek 722se sd on it
..like I say ,I'll look when I get home