iNavFlight/inav #10371: SKYSTARSF405HD4 Change timer for S5 S6

Yarolsav180703, 2024-09-19:
## Current Behavior
I use SKYSTARSF405HD4 FC and want to use 4 motors(S1-S4) and 2 servos (S5-S6) But its badly assigned to Timers like
S1 Timer4
S2 Timer4
S3 Timer3
S4 Timer2
S5 Timer3
S6 Timer3
And I can't use it like I want cause Timer 3 is already used by the Motor output 
## Desired Behavior
I asking for help to change the Timer in FC target, but I don't understand how to do it right, cause channels for alternative timers have an N index

## Suggested Solution
After googling I have seen some similar solutions, but I have Inav 7.1.1 and it seems the syntax of the Timer definition little bit other
I suggest changing Timer 3 to Timer 8 or Timer 1 with the N index

I write it in target.c, Timer 8 appears in Inav for S5 S6 S7(from where?) and servos don't working. Maybe I am wrong in syntax

#if defined(SKYSTARSF405HD2)
    DEF_TIM(TIM8, CH2, PB0, TIMER_OUTPUT_N_CHANNEL | TIM_USE_OUTPUT_AUTO , 0, 0),
    DEF_TIM(TIM8, CH3, PB1, TIMER_OUTPUT_N_CHANNEL | TIM_USE_OUTPUT_AUTO , 0, 0),
#endif

## Who does this impact? Who is this for?
All who use that FC and want wider functional

## Additional context
[image]

Similar request, but here used for 6 motors, I need 4 motors and 2 servos #9415

mmosca, 2024-09-19:
Not all timers are available to all pins. This is a hardware limitation.

Have you checked if there are other timers available for those pins? And have you tested if the new configuration still supports dshot correctly?

Yarolsav180703, 2024-09-19:
Thank you for your answer!
[image]
According to the datasheet, at those pins available Timer 1 and Timer 8 with index N, ive seen that its possible to use them with target tweaks like here: #3921
 
I tested, seems to DSHOT working correct, but servo outs 5-6 not shows any lifesigns
Also Timer 8 set to LED output, maybe its playing some role here

Maybe, I make uncorrect changes in target.c,

sensei-hacker, 2025-04-03:
> hey, any updates with this issue? I have same problems with this controller)

Why not just put the servos on S1 and S2, the motors on S3-S6?  Is there an AIO version of this board that has onboard ESCs on the first four outputs or something?

The version I'm finding is for external ESCs, so you can put the servos on S1 and S2. (Set timer 4 to servos).