iNavFlight/inav #10373: Add one touch loop feature

capblack, 2024-09-20:
Feature Request: Add one touch loop feature like Avata 2 or other quads, could be interesting

MrD-RC, 2024-09-20:
Sorry, this shouldn't need to be a "feature". It should be called being a pilot. Loops should be flown, not automated.

closed 2024-09-20 by MrD-RC

capblack, 2024-09-21:
Well, I don't share your opinion, since INAV itself includes many automation features, we could say the same about assisted takeoff and landing, waypoint navigation to give an example, so what would be the point if all these features had to be flown by the pilot as you say, it's just your point of view.

MrD-RC, 2024-09-21:
- Autolaunch adds assistance for tricky to launch aircraft. Some of which could need 2 hands to launch.
- Waypoint missions can be used where precision is needed. For example mapping missions

However, a "flip" mode is a gimmick. Its only purpose is because the pilot doesn't want to improve their skills. INAV shouldn't replicate the features of cheep Cheerson drones of the past. The fact the Avata 2 has it, just shows that it is a toy, not for pilots.

capblack, 2024-09-21:
Again it is your point of view, and I could say that it is not at all accurate to say that DJI Avata is a toy. I have a DJI Avata and its flight is almost perfect, INAV is at least half the precision of DJI, not to mention the camera, transmitter and goggles, it has an excellent camera, really a job of professionals as a collaborator of INAV I can understand that you are a little in love with it. I myself have 8 drones with INAV because it is something more like done by oneself. I started with the Rabbit controller and I know what it is to do everything by oneself, but you should not exaggerate and see a little more the reality as it is, DJI was and will be for another time the king.