iNavFlight/inav #10376: F4 1S 12A AIO Brushless Flight Controller V3

CorvusMonitoring, 2024-09-22:
This board is full of features for a very light design (4.2g), as an integrated serial elrs 2.4 rx, 2 serial ports, 12A esc, baro and efficient MCU, it would be interresting to integrate it to Inav project.

Specifically, we are developping a custom indoor drone for our dronepark. We need a light solution that works with 
2S li-ion (prototype has 15 minutes flight time on a CL25 frame), and full security for beginner with a MTF-01 Lidar/Optical flow for a static machine when stick released. Actually we've chosen Goku F745 nano stack, but it's expensive, heaviest, and not as easy as the Betaflight AIO...

If I can do anything for help, even supply a board for testing, I'll do!

Thank you,

Julien from Corbus Monitoring/Dronepark Geneva

techpolymath, 2024-09-23:
If I'm not mistaken this is the same board in the BetaFlight Pavo Pico. I would love to get INAV installed onto that board.  
MCU = STM32F405 correct?

CorvusMonitoring, 2024-09-24:
Yes, it is the STM32f405 version.