iNavFlight/inav #10380: BEEPER -ON_USB not working

mrduncan626, 2024-09-23:
I've been setting BEEPER -ON_USB via the CLI for many years but on the latest few versions of iNav, the beeper continues to beep while the FC is only plugged in the USB cable.  This issue occurs on all my Matek F405 Wing configurations.  But, this issues does not occur on my SpeedyBee F405 Wing configurations.  All of my FC’s are flashed with iNav 7.1.2 and using the 7.1.2 Configurator.

I've reached out several times on the iNav Fixed Wing FB group but with little to no responses.  Someone responded that he configurated his beeper to not beep when only plugged into the USD cable via the Configurator.  I've been using the Configurator for years and don't recall this feature was controllable from the Configurator.

Please Advise,
Thanks Mike

Jetrell, 2024-09-23:
Without seeing your beeper settings... There is a link between `beeper RX_SET` and `beeper MULTI_BEEPS` depending on the GNSS status.  Make sure both of these are enabled in your setting. Without them it can override `beeper -ON_USB` on some FCs I've tested, when powered by USB.

mrduncan626, 2024-09-24:
Everything is enable except for ON_USB.  see attached
[image]

Jetrell, 2024-09-24:
If it helps. These are the beeper settings I have on my Matek F405 wing. I haven't had an issue with it beeping when connected to USB, even with firmware up until now with an 8.0 master build... It maybe useful to go through them. 

[image]

Jetrell, 2024-09-24:
If it helps. These are the beeper settings I have on my Matek F405 wing. I haven't had an issue with it beeping when connected to USB, even with firmware up until now with an 8.0 master build... It maybe useful to go through them. 

[image]

Another thing to look at, relating to the GNSS module as I mentioned above.. Do you have it powered by a USB 4.5v supply. Or the 5v supply that requires the flight battery to be connected... If its the latter. And the software expects to see it, and does not, could also be a cause. 
There has always been some strange interaction between GNSS fix/ No fix, RX_set, multi_beeps, Dshot beeper and a Stand alone beeper.

mrduncan626, 2024-09-24:
The USB is connected to my computer.  The ESC is just a standard plane ESC and not DShot.  All I can tell you is up until a few iNav version before 7.0, the BEEPER -ON_USB worked and on my SpeedyBee FC's the BEEPER -ON_USB works in v7.1.2.

The Matek F405 have been a workhorse for iNav since the beginning.  I'll go back to 6.0.1 and see if it will work on that version and let you know.

mrduncan626, 2024-09-24:
Ok.  I just verified that in v6.1.0,  the beeper -USB_ON does turn off the beeper when be powered by my computer USD plug.

mrduncan626, 2024-09-24:
Here's the problem.  All my FC have a GPS connected to them.  The Matek F405 Wing does not power the GPS via the USB connection.  FC's like the SpeedyBee F405 WIng's, the USB connection provides power the GPS.

In v6.1.0, when connected to USB power, the firmware ignores the beeper warning that the GPS is not connected.  Hence the beeper -ON_USB turns off the no GPS beeper warning.

In later versions like v7.1.2, the firmware does not ignore the no GPS warning via the beeper even though the beeper -ON_USB feature is enabled.