iNavFlight/inav #10390: Enable Power and Amp Draw limits for Fixed wings

ltrzcinski, 2024-09-27:
## Current Behavior
For fixed wings, the maximum current/power output cannot be set other way than thru throttle scaling which is not optimal solution. 
See more info here: https://www.facebook.com/share/p/Utst9djnbSyixrqD/

## Desired Behavior
Enable following features and other related parameters to work with fixed wing configuration:
limit_burst_power
limit_cont_power
limit_burst_current
limit_cont_current

## Suggested Solution
Suggested solution is to either enable or provide user manual (if it is working already) to above features for fixed wings.

## Who does this impact? Who is this for?
For fixed wing pilots to safely limit motor loading regardless of prop and battery used. To prevent motor overheating and reduce the risk of inflight motor failure.
https://www.facebook.com/share/v/sf1ZcLWywBHKBwvf/

## Additional context
None.

MrD-RC, 2024-09-28:
The feature you mentioned are already in INAV. They have been for a long time. You can also limit current directly on BLHeli/Bluejay ESCs. 

- https://github.com/iNavFlight/inav/blob/master/docs/Settings.md#limit_burst_current
- https://github.com/iNavFlight/inav/blob/master/docs/Settings.md#limit_burst_current_falldown_time
- https://github.com/iNavFlight/inav/blob/master/docs/Settings.md#limit_burst_current_time
- https://github.com/iNavFlight/inav/blob/master/docs/Settings.md#limit_burst_power
- https://github.com/iNavFlight/inav/blob/master/docs/Settings.md#limit_burst_power_falldown_time
- https://github.com/iNavFlight/inav/blob/master/docs/Settings.md#limit_burst_power_time
- https://github.com/iNavFlight/inav/blob/master/docs/Settings.md#limit_cont_current
- https://github.com/iNavFlight/inav/blob/master/docs/Settings.md#limit_cont_power

closed 2024-09-28 by MrD-RC

ltrzcinski, 2024-09-28:
Hi @MrD-RC

MrD-RC, 2024-09-28:
There is nothing in the code that limits these parameters to specific platforms. They're global.

ltrzcinski, 2024-09-28:
Darren one question: if this is a global parameter, can it be that somehow global parameter (this or in general)  is disabled by default in inav? I know it sounds weird but thru the series of experiments I did with these settings, I never got it to work. Can be my ignorance, yes but just want to exclude other options.

sensei-hacker, 2024-09-28:
@ltrzcinski 
Does your log show the current changing, at reasonable levels? Low current at idle, then increasing at launch?

Have you tested setting it to say 2-3 amps to make it very obvious?

Also, it may be best to test in Angle, Acro, or manual modes rather than Autolaunch. If you have Autolaunch set with 90% throttle, and the current limit set to match 20% throttle, INAV could reasonably just look at you like you're crazy. :)

ltrzcinski, 2024-09-28:
Hi @sensei-hacker 
I tested it numerous ways. Have logs. Current is changing very normally. Current values are close to ecalc estimations. I tested max amp draw on bench and wanted to haircut few amps to give my sunnysky v2 longer life. While primarily I tested these settings in a wrong way ( =on bench) later I tested by simulating hand launch with WOT in angle mode without AutoLaunch. On this basis I collected logs  and numbers. As this is a global parameter as per @MrD-RC comment, I assume even if  90% throttle is set in autolaunch, it should be 90% of motor power limited by mentioned parameters. If autolauch overrides these settings it can end bad. Despite of what's written above, I do not exclude user (mine) error. I just do not know where to look.