iNavFlight/inav #10406: need firmware support for OMNIBUSF4V3

sujal141005, 2024-10-07:
Current Behavior
I'm using an AliExpress F4 flight controller with the ICM-42688 gyro. The flight controlle when flashed wth the OMNIBUSF4V3 does not show any gyro movement in inav. After modifying the code to switch from the BMI270 and MPU-6500 to the ICM-42688, I still encountered build errors when compiling for the omnibusf4v3 target, I'm really a beginner.

Desired Behavior
I would like support for the ICM-42688 gyro sensor to be added for the omnibusf4v3 target.

Suggested Solution
Add proper support for the ICM-42688 gyro in the firmware for this flight controller.

Who does this impact? Who is this for?
This is for users with flight controllers using the ICM-42688 gyro and also whoever is trying to use this or similar fc for planes.

Additional context
I'm using an AliExpress F4 flight controller with the ICM-42688 gyro. Link to the flight controller: [https://www.aliexpress.com/item/1005007354050504.html?spm=a2g0o.cart.0.0.6e8938daC6wFCO&mp=1#nav-specification].

sensei-hacker, 2024-10-07:
I am hesitant to start making changes to the Airbot Omnibus target, which works perfectly well for Omnibus boards, in order to have it instead run on a completely different board from a different manufacturer.

sujal141005, 2024-10-08:
I understand where you're coming from but I really just want help creating a custom target sicne you have the build environment ready just like it says in the docs. And that once a target header is made future people can just find this issue and fix their problem without a new board needing to be added to inav source code. I would be very grateful if you could help me out with this. everything other than the gyro seems to be working fine.

sensei-hacker, 2024-10-08:
You can unzip the attached file and test it.  It's not in any way an official INAV target.
It hasn't been tested - by me or anyone else. You can volunteer to test it if you wish.
It may fly very well for you. It may fly upside-down. It may fly away.

[inav_7.1.2_ALIFAKEOMNIBUSV3S.zip](https://github.com/user-attachments/files/17299839/inav_7.1.2_ALIFAKEOMNIBUSV3S.zip)

sujal141005, 2024-10-09:
you're a life saver! everything seems to be working so far, thank you so much! i will have to test it on ground, will update if anything goes wrong.

kriszbon, 2024-12-02:
Hello!
It's good that I found this site, because I have a similar problem (Gyro/Accelero was missing).
The linked Firmware solved it (it became ICM42605).
But he didn't do it in order when calibrating.
Yaw had to be rotated by -90 degrees in the Alignment menu, so it was set to approx. well.
Question, can this be good?
Anyway, if I tilted it to the right, it was Pitch up, when I tilted it to the left, Pitch down, and vice versa.
However, diagonal and many other movements were confusing.
Anyway, thanks for the effort.
[image]

kriszbon, 2024-12-02:
What I mean by this is that when rotated, everything moves in the right direction in the configurator (7.1.2).
Flight not attempted.

sensei-hacker, 2024-12-02:
> What I mean by this is that when rotated, everything moves in the right direction in the configurator (7.1.2). Flight not attempted.

Yeah that's fine.  The accelerometer on this board isn't in the same alignment as a genuine Airbot Omnibus, but that's no problem as long as the 3D model moves correctly.

kriszbon, 2024-12-02:
> > Ez alatt azt értem, hogy elforgatva minden a megfelelő irányba mozog a konfigurátorban (7.1.2). Nem kísérelték meg a repülést.
> 
> Igen, ez rendben van. Ezen a táblán a gyorsulásmérő nincs ugyanabban az állásban, mint egy eredeti Airbot Omnibusé, de ez nem probléma, amíg a 3D-s modell megfelelően mozog.

Hi!
Ok, thanks for your help.
Greetings.

kriszbon, 2024-12-06:
Hello!
Again, I would like help with the F4V3S case.
I don't know where to connect the IRC Tramp wire?
Uart1 port GPS, UART6 Rx, Uart3 Scl/Sda Magneto.
There is another free Tx3 pad here (maybe here?).
Is the magneto important during iNAV?
What would you miss if it wasn't connected?
Sorry, I'm not very familiar with these.
[image]
[image]

sensei-hacker, 2024-12-06:
Depending on what kind of receiver you're using, you clmaynbe able to put it on rx6 (and not get telemetry), then turn on soft serial on rx6 for Tramp.

That may not work, though. You may not be using Tramp on that board.

The mag is important for quadcopters. It is less important for airplanes, because we always know more or less which direction an airplane is flying - forward.

kriszbon, 2024-12-06:
Thank you for your answer.
This is a discounted machine, so Flysky FS-X6B PWM, PPM, I.BUS.
And Flysky FS-i6/10ch.
I use I,BUST in FPV mode.
This is Vtx.
OK, I'll delete what you wrote then.
Anyway, the IRC would be soldered to the + Tx3 pad.
[image]

kriszbon, 2024-12-06:
> Attól függően, hogy milyen vevőt használ, ráteheti az rx6-ra (és nem kap telemetriát), majd bekapcsolhatja a soft serialt az rx6-on a Tramp számára.
> 
> Ez azonban nem biztos, hogy működik. Lehet, hogy nem Trampot használsz azon a táblán.
> 
> A mag fontos a quadcopterek számára. Repülőgépeknél ez kevésbé fontos, mert többé-kevésbé mindig tudjuk, hogy egy repülőgép melyik irányba repül – előre.

Hi!
I have made the suggested change.
And yes, you can bring up the menu on the remote control (Pid, Vtx, Power, etc. etc...).
Now I just looked in the menus, I only took the Vtx down to 25mW.
Thanks again for your help.

closed 2024-12-11 by sensei-hacker