iNavFlight/inav #10409: Request to submit a new target configuration

HGLRC-T, 2024-10-14:
## Current Behavior

I've finished writing the configuration file and testing the firmware, HGLRCF405V2, and I'm requesting that my configuration file be merged

## Desired Behavior

Request to merge code

## Suggested Solution

## Who does this impact? Who is this for?

This allows users to directly obtain the firmware

## Additional context

closed 2024-10-14 by HGLRC-T

MrD-RC, 2024-10-14:
We welcome new firmware targets. Please note our new hardware policies.

https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

mmosca, 2024-10-14:
@HGLRC-T If you need help organizing shipping of the samples to developers, let me know. You can submit your target as a pull request and start the process from there.

HGLRC-T, 2024-10-16:
@MrD-RC Thank you! I have read it, I`m new menber of the team, so I tried to do the right way.
@mmosca Yes I need send some samples to the developers. And now I contack with SENSEI_HACKER by Discoard