iNavFlight/inav #10415: Inav Telemetry to Meshtastic over UART/Serial.

rotorHead82, 2024-10-16:
## Current Behavior
Inav currently can talk to certain SIM/sms modules to send telemetry and recieve commands through SMS messages. 
Having the same functinality by using a Mesthastic node instead would be great. It provides a great backup link for tracking and added security to be able to activate RTH for example should your TX loose connection. 
I already use a Meshtastic node with a GNSS module as a backup tracker through my own Meshtastic network in the area I fly. In case of a crash the node will provide easy location of the wreck as it is a standalone unit that run on its own battery and has its own gnss module. When airborne the range for a Meshtastic node is easily 100s of km if line of sight. (300km personal best from a com. Airplane.)
The programming for this should be fairly easy as all that needs to be sendt from the FC to the Node is a simple string with the telemetry data. (Ie: The text message the SMS/Sim module sends) The Meshtastic node can be configured to pass this through to the Meshtastic network as a message and oposite for recieved messages. 

## Desired Behavior
Having Inav send a string at configurable interval with Telemetry data through a UART connected to a Mesthastic Node.
Secondary, Have Inav react to receved messages/strings coming from the Mesthastic node activating functions. Examples: RTH, RTH_Land, Change altitude etc.

## Suggested Solution
Modify the already existent SIM/SMS function to only send a simple string of telemetry data. No AT commands or module control needed. The Meshtastic node will take this sting and spit it out as a message to the network.
Having the option for a "codeword" needed in a recieved message for Inav to react to it provides tamper proofing should the Meshtasic node not be set up for a private connection.

## Who does this impact? Who is this for?
Anyone using Mesthastic nodes or have a Mesthastic Network where they fly. Anyone who wants a backup tracking and control system. The transmitted packages will be retransmitted through anyones nodes so a private network is not needed for good coverage. If the Meshtastic node have its own GPS and battery it will also act as a standalone tracker and send positiondata even after a crash or the FC is disabled.

## Additional context
https://meshtastic.org/
https://meshtastic.org/docs/configuration/module/serial/
 Meshtastic serial Module config alternative:  -TEXTMSG- will allow you to send a string over the serial port to the *meshtastic*device, which will be broadcasted as a text message to the default channel. Any text message received from the mesh will be sent to the serial port as follows: <Short Name>: <text>.

xznhj8129, 2024-10-24:
can't you already do this? i mean, maybe the lora module can't have additional code on it, but i don't think it would be very hard to add a tiny ESP32 or Arduino that is connected to the telem serial port, just reads mavlink or LTM, takes the position and sends it to the meshtastic node by serial

MrD-RC, 2024-10-24:
> can't you already do this? i mean, maybe the lora module can't have additional code on it, but i don't think it would be very hard to add a tiny ESP32 or Arduino that is connected to the telem serial port, just reads mavlink or LTM, takes the position and sends it to the meshtastic node by serial

Or even better, MSP. That is the main comms protocol.

rotorHead82, 2024-10-24:
Having additional hardware to act as a translator/convertor in this case is a bit overkill and not a very delicate solution when we can have a direct serial link. Most of the functions to have inav commiunicate directly to the Meshtastic node is already in the SMS/GSM module code, it only needs to be simplified to serial write what would be the SMS text straight out on the serial port. We dont need all the AT commands and timings that the GSM module uses. Unfortually doing this is above my coding skills.
The Meshtastic node listens to the serial port and will prosses what is beeing dumped into it into a message for the Mesh network by itself. Unfortually Meshtastic does not have the bandwith for full continous telemetry transmittion.
To put it very simply we need inav to serial write the text that would be the SMS when using the SMS/GSM function, and be able to set how often it would transmit it (at minimum every 30seconds to not jam up the network.) The transmit interval function is already in the code as well as compiling the telemetry data into a text string.
Adding sending messages to Inav would be a "Nice to have" feature. This is also in the SMS/GSM feature.