iNavFlight/inav #10418: can't see the rc control mission number

Todor319, 2024-10-18:
controller speedyby wing ivav 7.1.2 . I receive ibus telemetri but I can't activate a sensor to show me a wp mission number

sensei-hacker, 2024-10-18:
As far as I can see, IBUS doesn't have a waypoint number field. LTM does.

https://github.com/iNavFlight/inav/blob/master/docs/Telemetry.md#available-sensors

You can, of course, display the waypoint number on your OSD. Unless you have a DJI, in which case you only get the values they decided you want.

Todor319, 2024-10-18:
can you tell me how i can show the point number

Todor319, 2024-10-20:
[image]
[image]
[image]
[image]
hi these are the sensors i am getting

MrD-RC, 2024-10-20:
Waypoint number is not sent via any control link telemetry. It is shown in the OSD. 

If you are doing autonomous missions without OSD. You should really use a ground station with a long range telemetry radio, such as mLRS, via MSP or MAVLink. They will have full access to mission information.

closed 2024-10-20 by MrD-RC

Todor319, 2024-10-20:
I am new to inav and I apologize if my inputs are wrong, I thought that the function is available for all controllers because I have purchased a controller with inav 4.1.0 installed, which through telemetry transmits information about mission points, which is quite convenient for rover
[image]