iNavFlight/inav #10428: Change flight mode violently changes height

Normoa, 2024-10-24:
I'm an (archaic) Inav user. Most of my quads are on Inav 2 or 3, and have always enjoyed Inav.

I'm setting up another quad, with different FC (now SpeedyBee F4 V3), and am trying out Inav 7.
I've got it flying in horizon mode OK, but when I switched it to PosHold with AltHold (it has a GPS and Barometer), it drops a lot of height and then recovers.

I suspect some of what I'm used to has changed, and/or I've got some settings wrong, but can't see anything that might aid in solving the problem.

Can anyone put me straight?

sensei-hacker, 2024-10-24:
Sounds like you need to increase the hover throttle setting.
When you first enter any altitude holding mode (including position mode / position hold), the first thing that happens is it sets the throttle to whatever you have set as your hover throttle. Then it starts adjusting to maintain vertical position.

Normoa, 2024-10-24:
Sounds sensible! 
I'll give it a try.

Hmmm... might be sensible for it to adopt the throttle setting that is currently being used..

sensei-hacker, 2024-10-24:
Yeah going with the current throttle seems logical at first. Complexities arise, such that there's no great answer.

closed 2024-10-24 by sensei-hacker