iNavFlight/inav #10434: Failsafe not working.

Lalbabu2001, 2024-10-27:
I setup my receiver as to send 850 value while radio signal lost to trigger failsafe rth.
 I found that rather than going into failsafe rth quad stops and fell while I turn off radio.
In osd it says disarmed by switch.
Why?

breadoven, 2024-10-27:
Which channel was set up to send 850 on Failsafe ? Hopefully not the arming channel because if that's the case it will just disarm when the Transmitter is switched off, with the disarm triggering before the Failsafe has time to activate.

As @sensei-hacker, said, without knowing what RC system this is and how it's been set up exactly it's impossible to know what you've done.

b14ckyy, 2024-12-05:
Closing as no update given by OP.

closed 2024-12-05 by b14ckyy