iNavFlight/inav #10436: accelerometer is not working accurately.

Lalbabu2001, 2024-10-28:
Hi, while reviewing my log i found that the accelerometer is not detecting slow reactions of quadcopter but it detects fast reactions only. as in the given attachment screenshot it its clearly seen during landing that baro detects decrease in altitude but accelerometer (acc z) is not detecting motion. Is it due to filters or any thing else?
[image]

sensei-hacker, 2024-10-28:
An accelerometer is not a motionometer. It is not supposed to detect motion.

It detects acceleration, CHANGES in motion. You can see at the end of the chart it very much detected when the motion stopped, when there was a change in the motion.

This is like if you're sitting in an airplane, cruising along. You might fall asleep. While the plane is going 500 MPH. You can't tell that it's moving - and neither can the accelerometer. You, and the accelerometer, notice at takeoff when it goes from not moving to suddenly moving very fast down the runway.

sensei-hacker, 2024-10-28:
Here we see the accelerometer dips when it reads the change from hover to descending. Once it's in a steady descent, there is no more acceleration for the accelerometer to report.
[image]

closed 2024-10-28 by sensei-hacker

sensei-hacker, 2024-10-28:
An accelerometer is not a motionometer. It is not supposed to detect motion, such as a steady descent.

It detects acceleration, CHANGES in motion. You can see at the end of the chart it very much detected when the motion stopped, when there was a change in the motion.

This is like if you're sitting in an airplane, cruising along. You might fall asleep. While the plane is going 500 MPH. You can't tell that it's moving - and neither can the accelerometer. The accelerometer in your inner ear notices at takeoff when it goes from not moving to suddenly moving very fast down the runway.

The accelerometers in your inner ear would notice if the pilot slammed on the brakes, creating a significant CHANGE in the motion. Steady motion can't be detected by an accelerometer. (Mainly because your craft isn't actually moving how you think - remember the earth is spinning at 1,000 MPH, so if it sensed motion it would detect that the aircraft is going 1,000 MPH 😁. And everything on earth is going 67,000 MPH around the Sun, so your drone is going 68,000. If you were "descending" in the morning, the drone was actually going up, but the earth was catching up to it. Yeah motion is weird.

anti-vaxxer, 2024-10-28:
@sensei-hacker Rigid body rotation and translation prove that people on a globe earth rotating and translating(orbiting the sun) would experience constant acceleration and deceleration. In reality we don't feel the earth beneath us accelerating or decelerating which disproves the globe model.
 Rigid Body Rotation With Translation Physics https://www.youtube.com/watch?v=i8BIzW5MKPA
 Physics 11.1 Rigid Body Rotation (1 of 10) Basics https://www.youtube.com/watch?v=suaJz-zWt3A

MrD-RC, 2024-10-28:
Sorry, but that's flat earther nonsense. We don't feel acceleration or deceleration because the earth is in free fall in its orbit around the Sun. Also, the order of magnitude of accelerations is tiny. Around 0.03 m/s2 at the equator, and 0.006 m/s2 due to the rotation around the sun. 0.03 m/s2 is the limit to the order of magnitude that humans can feel acceleration. And as pretty much everything around us is moving at the same velocities and accelerations, we can't sense the accelerations.

sensei-hacker, 2024-10-28:
> @sensei-hacker Rigid body rotation and translation prove that people on a globe earth rotating and translating(orbiting the sun) would experience constant acceleration and deceleration. 

If you're ever in Dallas, ping me. We can do some fun things things, including using the telescope to look at some interesting things.