iNavFlight/inav #10437: Set SafeHome point on the fly

DanielOBenson, 2024-10-28:
## Current Behavior
Safehome (or waypoint mission) is necessary for the autoland to work.  At the present time it's necessary to manually enter safehome points in the cli.

## Desired Behavior
It would be great to record a safe home point while flying over it.  Just like home reset works now.

## Suggested Solution
Enable recording a safe home point using a switch.   

## Who does this impact? Who is this for?
This would impact people who are flying any aircraft and who want to use the autoland feature.

## Additional context
THANK YOU!

MrD-RC, 2024-10-28:
Safe homes are not designed to be used like that. They are supposed to be nearby to the home point. So that you can physically get to the safe home and ensure that it is safe for your aircraft to land. I.E, there are no people there who may get hit by the aircraft. This system is not intended for landing away from home.

closed 2024-10-28 by MrD-RC

stronnag, 2024-10-28:
It is also possible to plan safehomes in both the INAV Configurator
[image]
and mwp
[image]
so there is absolutely no need to either edit a CLI diff or have an in-flight method.

DanielOBenson, 2024-10-28:
THANK YOU!

image.png (view on web)<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_user-2Dattachments_assets_329550dc-2D6ecf-2D444a-2D91fb-2D41b386015e67&d=DwMCaQ&c=euGZstcaTDllvimEN8b7jXrwqOf-v5A_CdpgnVfiiMM&r=788eX4Dw24pN30wom91CFTjVQtUIcuwoQr9EqYTYMnA&m=K4aBQo0wi35hCgcBiDOsfba6_FY93kdnPP7a4B5rg6QLh1gNybNnUvfg9H1svx55&s=pOmq_2ELtJiz9nVJVszusHfU65Bx7BKOyc2V96h1KTI&e=>
Screenshot.From.2024-10-28.20-18-23.png (view on web)<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_user-2Dattachments_assets_52053618-2D4d0e-2D4446-2Dbee5-2Dcf8981e70dac&d=DwMCaQ&c=euGZstcaTDllvimEN8b7jXrwqOf-v5A_CdpgnVfiiMM&r=788eX4Dw24pN30wom91CFTjVQtUIcuwoQr9EqYTYMnA&m=K4aBQo0wi35hCgcBiDOsfba6_FY93kdnPP7a4B5rg6QLh1gNybNnUvfg9H1svx55&s=VfHZrJCeRP7r5cwAKZpnUlj_4s-3OKb6u5CXeJ5GTKs&e=>