iNavFlight/inav #10442: GEPRC's GPS integrated barometer solution does not work

YI-BOYANG, 2024-10-31:
**Hello everyone, I am a staff member of GEPRC. Long ago, we found that there is no integrated barometer in the FC launched by some brands, which has no impact on the firmware of Betaflight. However, the lack of barometer will cause some functions of INAV firmware to be unable to be used normally.
We found this pain point, so have developed the GEP-M10 and GEP-M1025 GPS series, in the GPS integrated DPS310 or MS5611 barometer, just connect SCL and SDA can add to the flight control barometer, very popular with players, this is their product introduction link.**
* [GEP-M10](https://geprc.com/product/gep-m10-series-gps-module/)
* [GEP-M1025](https://geprc.com/product/gep-m1025-series-gps-module/)

****

**However, we found that the FC design of some brands is quite different, so that the barometer integrated on our GPS can not be activated normally in these FCS. Through the feedback of users, we summarized the following reasons:**

**1.The barometer definition is not added to target**
In most cases where the barometer cannot be turned on normally, we found that the FC target they used did not add the corresponding barometer definition, just add the definition of DPS310 or MS5611 to target.h, and then recompile the firmware to work properly.
`define USE_BARO_DPS310`
`define USE_BARO_MS5611`

**2.FC defines two I2C buses, but the barometer and magnetometer are not set on the same I2C bus**
Some FCS have 2 I2C buses, I2C1 for the external magnetometer and I2C2 for the internal FC barometer, e.g. GEPRCF405(this is a historical issue....) . You need to set the magnetometer and barometer to work on the same I2C bus, and then recompile the firmware to use the barometer integrated on GPS.
[image]

**3.The I2C address of the barometer conflicts**
The default I2C address of BMP280, DPS310, and SPL06 is generally 0x76. If the three models of barometers are integrated in the FC and the GPS integrated with DPS310 is connected to the FC, the I2C addresses of the FC and GPS barometers will conflict, resulting in the failure of the two barometers to work properly.

**We recommend that users have FC integrated with these barometric timings, which can be purchased with integrated MS5611 or GPS without a barometer. If you have already purchased a GPS integrated with DPS310, we offer two solutions:**

1. Turn off the barometer on your GPS
Remove the 0R resistor next to the barometer, you can turn off the power supply of the barometer, so that it no longer works, so that the I2C address conflict will not be caused, and the barometer integrated on the FC can be used normally. When it is necessary to reactivate the barometer on the GPS, it is only necessary to add a tin short at the position of the 0R resistance.
[image]

2. Change the I2C address of the DPS310 barometer on the GPS
If you must use the DPS310 barometer on the GPS, you need to remove a resistor next to the DPS310, so that the DPS310 I2C address will be changed to 0x77, so that there will be no I2C address conflict with the FC integrated barometer.
Then change the I2C address of DPS310 in common_hardware.c to 0x77, and then recompile the firmware for normal use.
`    BUSDEV_REGISTER_I2C(busdev_dps310,      DEVHW_DPS310,       DPS310_I2C_BUS,     0x77,               NONE,           DEVFLAGS_NONE, 0);`
[image]

**If you have other problems with GPS use, please mention them in this issue, thank you.**

thidas1290, 2024-11-21:
I'm using speedybee f405 mini FC and it's inbuilt DPS310 barometer is not working correctly. So I'm planning to use Geprc m10 dq gps module. if I understood correctly there's no software based method to disable inbuilt barometer, since it's on the same I2c bus and both have same address, therefor either I should remove the inbuilt barometer or should change the address of the DPS310 on gps module ?

YI-BOYANG, 2024-11-23:
> I'm using speedybee f405 mini FC and it's inbuilt DPS310 barometer is not working correctly. So I'm planning to use Geprc m10 dq gps module. if I understood correctly there's no software based [...]

If you can remove the 0402 resistor and change the INAV code and recompile the firmware, I think you can try the following to change the address of the DPS310.

James-Quigley-ie, 2024-11-26:
Thanks for the clarification. 

Thought I would add another option:
   • remove/desolder the onboard barometer 

I decided to do this for my SpeedyBee F7V3 as the DSP310 is said to be more accurate and I didn't want to recompile the firmware for every new version of iNAV or BF. For me, the new baro made a huge difference for alt hold mode. I would recommend 👌

thidas1290, 2024-11-26:
> Thanks for the clarification.
>
> Thought I would add another option: • remove/desolder the onboard barometer
>
> I decided to do this for my SpeedyBee F7V3 as the DSP310 is said to be more accurate and I [...]

I'm also planning to remove baro inside my f405 mini. May I ask how did you desolder the inbuilt one ? any aftermath pics

James-Quigley-ie, 2024-11-26:
Yeah sure. Below you can see the aftermath. Full temp on solder iron with a medium size solder tip. First the metal shield popped off, then after a minute of heating and prodding with fine tweezers the rest sort of fell apart. 

[image]

Schildkroet, 2024-12-26:
Hello,
i have a M1025-DQ and a Speedybee F405-Wing with onboard SPL006. With INAV6 and no Modifications, Baro was working, but it seemed the external (DPS310) was used. Maybe there was some crosstalk etc.

Removing 0R didn't work for me, external Baro was still responding. I had to remove 10K so the internal Baro was used.

YI-BOYANG, 2024-12-30:
> Hello, i have a M1025-DQ and a Speedybee F405-Wing with onboard SPL006. With INAV6 and no Modifications, Baro was working, but it seemed the external (DPS310) was used. Maybe there was some [...]

I'm guessing that removing 0R will actually show the built-in baro, since SPL06 will be recognized as DPS310.

Schildkroet, 2024-12-30:
> I'm guessing that removing 0R will actually show the built-in baro, since SPL06 will be recognized as DPS310.

I don't think so. I was blowing at the sensors to see which one is reacting in iNav. It was always the external one, which reacted. The I2C lines were probably providing enought power to run the sensor.

OWHt, 2025-03-13:
Hi, the barometer on my Speedybee f405 v4 has broken. As it's my long range drone, I'm thinking of replacing it with a DSP310 baro. Has anyone tried this combination with the M10-DQ/DI? 
Speedybee support couldn't help me with this.

YI-BOYANG, 2025-03-14:
> Hi, the barometer on my Speedybee f405 v4 has broken. As it's my long range drone, I'm thinking of replacing it with a DPS310 baro. Has anyone tried this combination with the M10-DQ/DI? Speedybee support couldn't help me with this.

I think so. But you need to make sure that the barometer on the FC doesn't interfere with the barometer on the GPS. Like removing the barometer from the flight control.

RandomNameF0, 2025-04-19:
@James-Quigley-ie@thidas1290 I removed the onboard barometer on my SpeedyBee F405 mini and now it's showing red barometer icon on Inav???

James-Quigley-ie, 2025-04-19:
@RandomNameF0  did you tell betaflight your new barometer is the DSP310 chip?

thidas1290, 2025-04-29:
@RandomNameF0 as far as I remember both units are using same barometer or interface so there was no build flags need to be added to the build. Did you properly wire up the i2c connections too ?, gps is using uart and the barometer and magnetometer is using i2c protocol, so u need to wire up the two wires too.

RandomNameF0, 2025-10-15:
@thidas1290 I just came back rebuilding my second drone and noticed this same module GPS I've been using dont have resistor from the factory?

[image]