iNavFlight/inav #10443: USB recogonized but Failed to open MSP connection on Debian 12 Linux

Kai2510, 2024-10-31:
Hello, I'm using INAV Configurator 7.1.1 on Debian 12 Linux. However, I can't connect to the flight controller while it works fine on Windows 10. The board is checked to be OK. The configurator runs well on linux in demo mode.

I haven't install drivers and don't know where to download Linux versions. Don't know if it is caused by driver problem.
```
[ 2431.301733] usb 3-2: new full-speed USB device number 12 using xhci_hcd
[ 2431.455119] usb 3-2: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
[ 2431.455124] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2431.455126] usb 3-2: Product: MatekF405TE_SD
[ 2431.455127] usb 3-2: Manufacturer: INAV
[ 2431.455128] usb 3-2: SerialNumber: 0x8000000
``
The serial port is `/dev/ttyACM0` by default. It can be opened using serial port debugging tools such as cutecom, but no data received while blue LED of the board flashing.

Any suggestion, hint and help will be appreciated. 
---

- FC Board name and vendor: MATEKF405TE_SD
- INAV version string:  OS: Linux, Chrome: 98.0.4758.82, Configurator: 7.1.1

sensei-hacker, 2024-10-31:
You won't need to install any drivers. Linux comes with drivers.

I see you have the journalctl output from when you plugged the board in. Check journalctl when you try to connect with Configurator. You'll probably see one of two things:

Permission error. Add the user to the "dialout" group and make sure that group has write permissions on the device.

Alternatively, the port may be grabbed by some other software you have running, such as the Arduino IDE or a 3D print slicer such as Cura.

Kai2510, 2024-10-31:
> You won't need to install any drivers. Linux comes with drivers.
>
> I see you have the journalctl output from when you plugged the board in. Check journalctl when you try to connect with Configurator. [...]

Thanks! It was "access denied" error. Adding the user to "dialout" group by ` sudo usermod -a -G  dialout user` works. After logout & login, it works well. Temporarily solve the problem by executing `chmod 766 /dev/ttyACM0` each time of connection also works.

closed 2024-10-31 by Kai2510