iNavFlight/inav #10445: Servo rotation on power-on

jhoppenstedt, 2024-11-01:
## Current Behavior

I have a fixed wing configuration with 1 motor (M1) and 2 CS-929MG analog servos (S3, S4).
When connecting the battery, the servo connected to S3 initializes and rotates ~90 degrees before returning to the "home" position where it was before powering on. The one connected to S4 just slightly wiggles. Switching servos also keeps the behaviour in relation to the port.

That 90° rotation will let the connected rod collide with the wing cutout though.

I also observed that the S3 servo will just wiggle sligtly on power-on, if the FC is already connected to USB (5V) before connecting the battery.
My conclusion is that the servo behaviour probably relates to instable voltages. It might also be a combination of voltage and model/manufacturer.

## Steps to Reproduce
1. Connect servos to S3 and S4
2. Power on FC
3. Observe S3 connected servo to init with a lot of rotation

## Expected behavior

I would expect the FC firmware to wait with peripheral init until voltage stabalizes.

## Suggested solution(s)

Another idea would be adding a radio control mapping allowing to disable the servo output (like "servo arming") that allows me to enable them after the FC powered up.

## Additional context

[iNav Config](https://pastebin.com/5vcu2Z8W)

---

- FC Board name and vendor: SpeedyBee F405 Wing
- INAV version string: INAV/SPEEDYBEEF405WING 7.1.2 Jun 1 2024 / 00:47:23 (4e1e59eb) GCC-10.3.1 20210824 (release)

b14ckyy, 2024-11-17:
The problem is, as far as I know, not on the INAV firmware but actually the Flight controller MCU itself. I see the behavior very differently on different FCs even of the same model. The MCU is leaking noise on the signal pads during the bootup process, causing this movement. When the Servo recenter, this is when INAV actually initializes the signal pads. There is nothing we can do about it.

jhoppenstedt, 2024-11-17:
Thank you for the thoughts! I'am now using S4 and S6 ports which don't seem to have that issue

closed 2024-11-17 by jhoppenstedt