iNavFlight/inav #10455: motors stop working while trying to take off

MohamedNour1452, 2024-11-05:
## Current Behavior
<! -- when I try to raise the drone up it motors just stops working momentary and start working again maybe after the drone is already flipped. it happened to my 3 times and the drone works perfectly fine after the crash and before as well. 
## Steps to Reproduce

1. arming the quad
2. raising the throttle 
3. airmode off

## Expected behavior

## Additional context

---
log: 
[LOG00046.TXT](https://github.com/user-attachments/files/17639396/LOG00046.TXT)
cli:
[INAV_7.1.2_cli_20241106_003329.txt](https://github.com/user-attachments/files/17639428/INAV_7.1.2_cli_20241106_003329.txt)

- speedybee f405 v4
- INAV version : 7.1.2

sensei-hacker, 2024-11-06:
What wasn't mentioned here is the user ID forcing the throttle to 0% using the programming tab.

MohamedNour1452, 2024-11-07:
Hello there, since you are the one replying here Let's just continue our discussion in discord

closed 2024-11-17 by b14ckyy