iNavFlight/inav #10456: OSD Freeze, Modes lost, no RTH

cooool39, 2024-11-06:
Hello,
i´ve got a huge probleme with my Setup (Matek H743 Wing).
in the middle of my flight inav (7.1.2) goes Crazy:

first i lost the direction Arrow
then the OSD freeze (Walksnail)
and the all Modes (angle, rth etc) are gone or different

i had full Stick control but no assistens (i could not activate RTH)

after 5-10min everything was normal, but then it starts again.

- INAV/MATEKH743 7.1.2 Jun  1 2024 / 00:04:49 (4e1e59eb)
- GCC-10.3.1 20210824 (release)

mmosca, 2024-11-06:
Do you have black box log of this? What is your radio system? 

Also, share you diff. What you shared is not really enough information for any investigation.

And I would suspect a potential hardware issue. Maybe the becs are strugling to keep voltage levels.

b14ckyy, 2024-11-17:
Will close for now as no further feedback was given.

closed 2024-11-17 by b14ckyy

Jetrell, 2024-11-17:
@b14ckyy I'd say he turn down the logging rate, or turn off the logging all together. And found it now works fine