iNavFlight/inav #10470: inav configurator 7.1.2 does not show reason for navigation is not safe (cli same)

swissembedded, 2024-11-16:
## Current Behavior
I nav does not arm. Arming disabled flags only show CLI in CLI.
Everything seems fine GPS 3d lock, receiver center, calibration, ...

Problem on speedybee F405Wing Mini and Matek F765 Wing. Both new setup with inav 7.1.2.

b14ckyy, 2024-11-16:
Added in 8.0 as a new feature: https://github.com/iNavFlight/inav/pull/10280
Also in Status bar: 
[image]

sensei-hacker, 2024-11-16:
The detail of why navigation is unsafe is shown on the OSD for 7.1.2.  To see it in the CLI, you can try the new INAV 8.0 RC1 that was just made available.

https://github.com/iNavFlight/inav/pull/10280

swissembedded, 2024-11-17:
I checked with osd, it just shows gps as reason, but in inav I see that it has 3d lock and values are fine.
[image]
[status.txt](https://github.com/user-attachments/files/17790105/status.txt)
[INAV_7.1.2_cli_20241117_110704.txt](https://github.com/user-attachments/files/17790106/INAV_7.1.2_cli_20241117_110704.txt)

Also tried to remove ground assistance type, galileo, beidou, did not help. HDOP, EPH, EPV seems fine to me. Magnetometer is disabled. It would be nice if it could show the reason in details. Will check 8.0 too and give feedback.

b14ckyy, 2024-11-17:
I see your problem. 

set inav_max_eph_epv = 1000

this should fix it. 
its a bug we have still not found the cause of. the value of this parameter is sometimes set to 1.000 instead of 1000. 
Change that in CLI and it should be fine again.

swissembedded, 2024-11-17:
Yes, that that was it, will test with the other if the same and with 8.0. Thank you very much.

sensei-hacker, 2024-11-17:
Was the configuration of the Matek board copied over from the SpeedyBee board?

swissembedded, 2024-11-17:
Speedybee board exactly the same issue. 

Matek: old config with inav 6.0 updated to 7.1.2, I did not do full erase, but flashed and used setup / reset setting and configured everything from scratch including calibration. Guess that is the mistake.
Speedybee: Virgin Board from speedybee, I think it had ardupilot installed (not sure anymore). Flashed and most probably did not select full erase either. Then configured everything from scratch.

b14ckyy, 2024-11-17:
Full erase is always recommended but does not cause this issue. 
Something is causing this setting to be set wrongly and we still don't know why. Most people reporting that are using SpeedyBee F405 Wing FCs but also some Matek ones. We don't even know if that happens on first initialization after flash or with a diff transfer or just randomly later during use... Its really annoying when it happens.

closed 2024-11-17 by b14ckyy

swissembedded, 2024-11-18:
I have  3 new controllers from speedy be with f405 left here to fool around for some old models to add inav. Any hint how to setup to reproduce?

b14ckyy, 2024-11-18:
We don't know how to exactly reproduce issue. 
You can try: 
- Check what INAV version they have out of box and what value is set there from factory
- set some basics like mixers modes and change a few settings. then pull a diff and look if that setting is in the DIFF and with what value
- Update INAV to 7.1.2 with full erase and maybe also once with no chip erase and then check in CLI the parameter again
- Then load back the previous diff and check again the parameter... 

Would be interesting if at any of these steps the value changes or when it is at 1.000 again in general.

mlochen, 2025-03-23:
> I see your problem.
> 
> set inav_max_eph_epv = 1000
> 
> this should fix it. its a bug we have still not found the cause of. the value of this parameter is sometimes set to 1.000 instead of 1000. Change that in CLI and it should be fine again.

Thank you so much for this solution! I was already doubting my sanity because all the sensors worked perfectly in inav-configurator but it still said that navigation wasn't safe and wouldn't let me arm. I have a SpeedyBee f405 wing mini with inav 7.1.2 on a flying wing if that helps.