iNavFlight/inav #10472: Inav 7.1.2 GPS issues with FlyFishRC M10 Mini GPS Module

roroid, 2024-11-18:
## Current Behavior
Not able to set GPS satelites to Galileo or Glonas
Not able to arm the vehicle, even if I have more than 10 satelites locked

https://github.com/user-attachments/assets/fa14cd08-8710-4172-8280-e9334e7f4776

## Steps to Reproduce
See above video

## Expected behavior
Able to setup Galileo or Glonas.
Arm with more then 6 satelites locked

## Suggested solution(s)
if I configure the variable
`gps_auto_config=off`
system is working
## Additional context
This is an old FC, Betaflight F4. I did not have this issue with a Speedeby Mini WING F405 on same INAV version with same type of GPS.
---

- FC Board name and vendor:Betaflight F4
- INAV version string:7.1.2

mmosca, 2024-11-18:
Please post the output of status and/or try 8.0.0 RC1

roroid, 2024-11-18:
`ntering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# status
INAV/BETAFLIGHTF4 7.1.2 May 31 2024 / 21:59:42 (4e1e59eb)
GCC-10.3.1 20210824 (release)
System Uptime: 188 seconds
Current Time: 2024-11-18T15:48:10.949+00:00
Voltage: 0.55V (0S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=MPU6000, ACC=MPU6000
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=NONE, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
ESC Temperature(s): Motor Count = 2
ESC 0: 0°C, ESC 1: 0°C, 
Stack size: 6144, Stack address: 0x10010000, Heap available: 1828
I2C Errors: 45, config size: 10423, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = none, used = none
   CURRENT : configured = ADC 1, used = ADC 1
  AIRSPEED : configured = none, used = none
System load: 9, cycle time: 1014, PID rate: 986, RX rate: 9, System rate: 9
Arming disabled flags: NAV RX CLI
OSD: MSP DisplayPort: BetaFlight Compatability mode (HD) [53 x 20]
VTX: not detected
GPS: HW Version: UBLOX10 Proto: 34.00 Baud: 115200
  GNSS Capabilities:
    GNSS Provider active/default
    GPS 1/1
    Galileo 1/1
    BeiDou 1/0
    Glonass 0/0
    Max concurrent: 3
`

sensei-hacker, 2024-11-19:
First, you'll want to use "UBLOX7" protocol. That's for N7M or newer. You can first check if just doing that solves your issue.

If not, the next thing is if you manually changed any settings on the GPS using U-Center, go back to U-Center and click the button to reset to defaults. The M10 is very limited hardware - only one single receiver front end, for example. So each channel can only be used by one constellation. If you have a different constellation set to use the channel that INAV is trying to configure for that channel, it'll be a conflict and not work.

roroid, 2024-11-19:
I don't think i follow you. I want to setup Galileo, and I'm not able because the settings are getting reset. I'm not trying to mix any settings.

imotep33, 2024-11-29:
Hello, i'm having exactly the same issue. I use this model of GPS on a speedybee F405 MINI wing flight controller. 
I can't set up the Galileo satellites ON. I only have the GPS satellites working.
I also tried to use Ublox or Ublox7 protocol and there is no difference. 

I have exactly the same GPS on a matek F411 WTE and Inav 6. it's working perfectly and the Galileo satellites are ON

sensei-hacker, 2024-11-30:
> I don't think i follow you. I want to setup Galileo, and I'm not able because the settings are getting reset. I'm not trying to mix any settings.

What exactly do you mean when you say ", I applied the M10 hack to it"?  What exactly did you do?

sensei-hacker, 2024-11-30:
> Hello, i'm having exactly the same issue. I use this model of GPS on a speedybee F405 MINI wing flight controller. I can't set up the Galileo satellites ON. I only have the GPS satellites working. I [...]

If you indeed have exactly the same problem, you tried to do some kind of "hack" to it? Probably using U-Center?

imotep33, 2024-11-30:
> If you indeed have exactly the same problem, you tried to do some kind of "hack" to it? Probably using U-Center?

hum, no. I don't know how to do it I a preferred to spinelessly downgrade to inav 6.1.1 and Galileo works fine on this version.

roroid, 2024-12-03:
> > I don't think i follow you. I want to setup Galileo, and I'm not able because the settings are getting reset. I'm not trying to mix any settings.
> 
> What exactly do you mean when you say ", I applied the M10 hack to it"? What exactly did you do?
What is writen here.

https://iflight-rc.eu/blogs/news/m10-gps-3d-fix-enhancement?gad_source=1&gclid=CjwKCAiA9bq6BhAKEiwAH6bqoCNzhX-oZ7AIqF6fMruQxLgygtqmLqYrekVkk1vZHI_tSTznM-pMlxoChKkQAvD_BwE

sensei-hacker, 2024-12-03:
Okay so you 

> What is writen here.
>
> https://iflight-rc.eu/blogs/news/m10-gps-3d-fix-enhancement?gad_source=1&gclid=CjwKCAiA9bq6BhAKEiwAH6bqoCNzhX-oZ7AIqF6fMruQxLgygtqmLqYrekVkk1vZHI_tSTznM-pMlxoChKkQAvD_BwE

Okay so you applied a bunch of settings from someone who writes " Some cryptic stuff should appear, not sure why but it does. .... Select every available checkbox option ... "  IN other words, they don't really know what they're doing, and clearly you don't know what you did.

Part of what's in those settings is it tells the module to update every 1 millisecond - 1,000 times per second.
The chip is actually capable of updating 10 times per second.

The suggestion remains the same - reset to defaults in UCenter to get the module back into a sane state, then let INAV configure it as required. Alternatively, you could restore the backup of the factory settings that you saved before making changes. Or maybe the person writing that post didn't know you should svae a backup before making changes?  To restore your module to a reasonablestate, go to View -> Configuration View, select “CFG (Configuration)” on the left, select “Revert to default configuration” on the right, and press “Send” at the bottom.

sensei-hacker, 2024-12-03:
> Still, to avoid losing track of where we began I’m still not able to select Galileo

And you probably won't be able to until you reset the module to a sane state.

roroid, 2024-12-03:
> > Still, to avoid losing track of where we began I’m still not able to select Galileo
> 
> 
> 
> And you probably won't be able to until you reset the module to a sane state.
> 
> 
> 
> 

Ok. Seems that I'm speaking American and you understand English.

imotep33, 2024-12-04:
there are 2 things here:

- One issue is about bee able to set the Galileo sats ON which was also my issue in Inav 7, and it's the video enclosed in the first post. the FC just refuses to set galileo ON. when you sent the command, and you check it back. it stays OFF.

- the second issue he is talking about is: not beeing able to arm, even with 12 sats and a steady fix. I also had it by the past on my speedybee FC. I saw it's a common bug, and I solved it by reflashing the card (It was inav 7 at that stage) and re configure from scratch. but it seem it can be solved by flashing + reset the parameters from the diff without the satellites parameters.

sensei-hacker, 2024-12-04:
> * the second issue he is talking about is: not beeing able to arm, even with 12 sats and a steady fix. I also had it by the past on my speedybee FC. I saw it's a common bug, and I solved it by [...]

The second issue you're referring to occurs when that Speedybee is upgraded from INAV 6 to INAV 7 with full erase turned off.  That causes theax EPH to be set to 1 centimeter, rather than 1,000 centimeters.

imotep33, 2024-12-07:
I had some time free on the bench, I upgraded my FC to Inav 8 RC2
Remember, my GPS is stock, i did not modify anything.
The issue still there

_______________________________________
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# get gps_ublox_use_galileo
gps_ublox_use_galileo = OFF
Allowed values: OFF, ON

# set gps_ublox_use_galileo = ON
gps_ublox_use_galileo set to ON
# get gps_ublox_use_galileo
gps_ublox_use_galileo = ON
Allowed values: OFF, ON

Save

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# get gps_ublox_use_galileo
gps_ublox_use_galileo = OFF
Allowed values: OFF, ON
____________________________________________

sensei-hacker, 2024-12-08:
Thanks for that info @imotep33 
I understand in your case you have NOT used UCenter to mess with any settings; have not loaded any ucf file from any blog post.

The module I would use to run a test to double-check with your exact configuration is broken at the moment, so we'll have to take a different path.  The first thing that can cause issues is that upgrading from INAV 6 to INAV 7 with "full chip erase" turned off sometimes messes up settings.  be sure to check that box whenever switching between major versions - 6 to 7, 7 to 8, etc.  The issue of "not beeing able to arm, even with 12 sats" is an example of that.

You can save a diff and copy settings over as described in the release notes (some settings carry over, others don't).
Can you please do a flash with full erase of either 8.0RC1 or 7.1.2, even if you've already done that before, just to double check? Then we can proceed from there.

imotep33, 2024-12-08:
hello, I did it again. with the full chip erase function ON. still got the issue with inav 7.1.2 and Inav 8 RC2

tizzism, 2024-12-18:
I have the SPeedybee Wing Mini. with Be-220 gps from Beitian (off amazon). normalgps baud in inav. I could not get my GPS to lock, with 13 satellites.

I changed the CLI Command: set inav_max_eph_epv = 1000
 > tested > worked. Fixed enabled. 

# status
INAV/SPEEDYBEEF405WING 7.1.2 Jun  1 2024 / 00:47:23 (4e1e59eb)
GCC-10.3.1 20210824 (release)
System Uptime: 629 seconds
Current Time: 2024-12-18T16:03:34.105+00:00
Voltage: 0.00V (0S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=ICM42605, ACC=ICM42605, BARO=SPL06
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: Startup failed
Stack size: 6144, Stack address: 0x10010000, Heap available: 1728
I2C Errors: 0, config size: 10467, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 4, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = ADC 3, used = none
System load: 5, cycle time: 1004, PID rate: 996, RX rate: 9, System rate: 9
Arming disabled flags: NAV RX CLI
OSD: MAX7456 [30 x 16]
VTX: not detected
GPS: HW Version: UBLOX10 Proto: 34.09 Baud: 115200
  GNSS Capabilities:
    GNSS Provider active/default
    GPS 1/1
    Galileo 1/1
    BeiDou 1/1
    Glonass 1/0
    Max concurrent: 3

[image]