iNavFlight/inav #10475: Allow to adjust gimbal sensitivity with radio input when using Serial control of the gimbal instead of PWM control in INAV 8.0

jaroszmm, 2024-11-20:
## Current Behavior
Right now, if using Serial Port to control gimbal movement and operation mode we can have assigned 3 radio channels for yaw/pitch/roll movement of gimbal plus 4th channel to select operation mode. 
However if using Serial Port to control there is no way to adjust sensitivity other than typing the value and have it fixed.

## Desired Behavior
Possibility to assign one radio channel to sensitivity adjustment, similar like it works with PWM version of gimbal control so it can be adjusted midflight to increase or decrease stabilisation

## Suggested Solution
Add option to assign sensitivity value to a specific radio channel similar how it works with yaw, pitch, roll now.

## Who does this impact? Who is this for?
I believe it's for anyone using the gimbal and wants to have more control over stabilisation.

## Additional context
The whole idea came from the fact that in FPV mode of the gimbal, when approaching landing or doing any precise movements any stabilisation is not needed. So in FPV mode it's better for the camera to be "locked". To achieve that now I'm using PWM control and some logic conditions in INAV so if I switch to Acro or Manual then in FPV mode sensitivity is maxed up resulting in the camera being practically not stabilised by the gimbal at all. 
But during a normal flight in Angle mode some stabilisation (not the full horizon lock one) is nice to smoothen the picture, to mitigate some plane movement. And then it's nice to be able to lower sensitivity value and stabilise the picture a bit.

However in serial control of the gimbal in 8.0 it seems sensitivity can have only one fixed value. It looks like the sensitivity adjustment works only for FPV mode of the gimbal anyway. In horizon lock (full stabilisation mode) changing sensitivity doesn't change anything regarding how the camera is stabilised.

glenn-venghaus, 2024-11-20:
There is already a tested PR that achieves this via the programming framework. 
I used it and works great. 
https://github.com/iNavFlight/inav/pull/10405

jaroszmm, 2024-11-20:
> There is already a tested PR that achieves this via the programming framework. I used it and works great. #10405 Unfortunately (not sure why as works great) it was moved from 8.0 to 8.1 release....

Thanks. I wonder if this can be simplified and moved directly to the configuration page or maybe made accessible in programming tab for an option like :
IF flightmode=acro or manual then set sensitivity to 15, ELSE set sensitivity to 5..... etc

glenn-venghaus, 2024-11-20:
It IS in the programming tab/framework. Thats what this PR is about. So there you can do exactly as your  mentioned example.  (it is linked to and requires the PR for the configurator to make this work from the UI or alternatively like i did use the cli untill it is properly merged into the master).
I have used it to drive it from an (scaled down) RC channel and also as an overwrite from a switch when i start landing to lock the sensitivity to the max.
I think the only thing that would make it even better is to also be able to drive it from the OSD as tunable parameter, but this work for me fine as it is.

jaroszmm, 2024-11-20:
Oh man, sorry, I just realised I confused the programming tab with adjustments tab. So this is great, some logic conditions to lock the camera in place in certain situations will be great. I see you have the same approach as me - lock camera for landings, as with stabilisation, even small, it makes approaches and flaring harder if landing FPV, not LOS.

glenn-venghaus, 2025-03-09:
fyi, that pull request has been merged now into the master. So if you dont mind flying development builds you can use the nightly builds of the configurator and inav and cabn use the programming framework as mentioned to drive the sensitivity with any rc chnnel (and some math) you like.
So this issue can be closed

jasc76, 2026-10-18:
Please close issue when its fixed