iNavFlight/inav #10487: After intalling MTF-01 optic flow althold does not work.

and-sh, 2024-11-29:
Inav 8.0.0
Althold was tuned before installing optflow. Worked normal.
After installing optflow.
Under approx 2m - work normal.
Higher does not work.
Testing in field over flat surface (snow). Sunny day.
Surface mode = OFF
nav_max_terrain_follow_alt = 100
inav_max_surface_altitude = 100

Max altitude where rangefinder measure reliable data 120cm.
[image]

Max altitude where rangefinder measure something about 3m.
[image]

About 16m rangefinder report zero altitude
[image]
At the same time surface reliability = 1000.
 
Log
[blackbox_log_2024-11-28_131957.TXT](https://github.com/user-attachments/files/17958253/blackbox_log_2024-11-28_131957.TXT)

and-sh, 2024-12-03:
Nobody need rangefinder here?

and-sh, 2024-12-03:
Is the way disable rangefinder from switch?

sensei-hacker, 2024-12-03:
> Is the way disable rangefinder from switch?

Turning off surface should turn off the rangefinder, except for autoland. Unless I'm mistaken.

and-sh, 2024-12-03:
I think I got it. The rangefinder readings are not used directly to calculate the position. When the quality of the readings is high, they are used to calculate the correction for the barometer and GPS, when the quality is low, this correction is simply a constant.

What is the difference between the following parameters?
nav_max_terrain_follow_alt
inav_max_surface_altitude

and-sh, 2024-12-03:
If I set to zeroes 
inav_w_z_surface_p
inav_w_z_surface_v
Will it effectively disable rangefinder for z position estimator?
Will surface mode work in this case?

sensei-hacker, 2024-12-03:
> If I set to zeroes
> inav_w_z_surface_p
> inav_w_z_surface_v
> Will it effectively disable rangefinder for z position estimator?
> Will surface mode work in this case?

Yes, that would disable the sensor that measures the distance to the surface. Which would make surface mode not work.

> What is the difference between the following parameters? nav_max_terrain_follow_alt inav_max_surface_altitude

They are almost the same.  max_terrain_follow_altitude is the maximum altitude it will go in surface mode. Your "throttle" (altitude)  input is scaled based on this value.
inav_max_surface_altitude is the maximum reading at which your rangefinder should be trusted.

It often makes sense to set them to the same value.

and-sh, 2024-12-04:
It seems to me that the equal sign should be removed in this place.
[image]
Because
[image]

and-sh, 2024-12-04:
If somebody can fast recompile corrected source for matek 722 mini se, I can give it a try

sensei-hacker, 2024-12-04:
It seems so to me as well. At very least, in the case of that particular type of sensor. I wonder if the same applies to others?

See also the similar line around 

https://github.com/iNavFlight/inav/blob/1c15fcc3deedbf0159968d33f6f2790f03665637/src/main/sensors/rangefinder.c#L250

Are you familiar with how to make a pull request?

sensei-hacker, 2024-12-04:
That rangefinder is connected via MSP, right?
So it should already be handled here
https://github.com/iNavFlight/inav/blob/1c15fcc3deedbf0159968d33f6f2790f03665637/src/main/io/rangefinder_msp.c#L68

Above, you wrote "About 16m rangefinder report zero altitude", but your screenshot shows 1 rather than 0.

and-sh, 2024-12-04:
> Above, you wrote "About 16m rangefinder report zero altitude", but your screenshot shows 1 rather than 0.

That means the problem is somewhere else.
Will try to check readings by rangefinder itself. Micoair have dedicated programm. May be this is problem of rangefinder itself.

and-sh, 2024-12-04:
I think I found the real problem. The rangefinder was installed incorrectly.

sensei-hacker, 2024-12-04:
Glad you found the problem!
For the next person, what was incorrect about the installation? Was there something below the rangefinder it was getting a reading of? That is, something "blocking" part of the view so it was reading the distance the landing gear or whatever?

and-sh, 2024-12-04:
There was nothing directly below it. But it has a field of diffusive light of almost 180 degrees. Any object that hits it scatters light and becomes decisive at high altitudes, when the main signal becomes weak. The sensitivity of this simple devise is amasing.

It is necessary to have a switch to turn off the rangefinder, since any scattering from dust or fog can disrupt stabilization and lead to a sharp change in altitude

and-sh, 2024-12-06:
The rangefinder started working
[image]

and-sh, 2024-12-08:
Ok. I remove the optical flow and return to the working combination of GPS + barometer + compass. If anyone in the future is interested in finishing the optical flow and rangefinder, contact me.

closed 2024-12-08 by and-sh

utrinque96, 2025-01-01:
Check and comment on this, please:
https://github.com/iNavFlight/inav/issues/10566#issuecomment-2566742928