iNavFlight/inav #10494: Empty Window in Wayland

6543, 2024-11-30:
If I start the `nwjs 124.0.6367.62` on an Linux v6.12.0 x86_64

I get an transparent window.

My DE is Hyprland v0.41.2

****

## Current Behavior

Nothing to see

## Expected behavior

I wana see the window with its features

## Suggested solution(s)

Use an other system?

## Additional context

This was used on an NixOS via: `nix shell nixpkgs#inav-configurator`

stronnag, 2024-11-30:
And what happens when:
* You use the official Configurator version from the INAV repo?
* You use a more conventional DE (xfce, Gnome Shell, kde)?

6543, 2024-11-30:
- using distrobox to run the oficial distributed [INAV-Configurator_linux_x64_8.0.0.deb](https://github.com/iNavFlight/inav-configurator/releases/download/8.0.0-RC1/INAV-Configurator_linux_x64_8.0.0.deb) it works and starts with xwayland ... so xorg mode

closed 2024-11-30 by 6543