iNavFlight/inav #10497: Inav 7.1.2 Heavy Horizon Drift - AHRS not working correctly

noahwaldner, 2024-12-02:
## Current Behavior

When flying my Delta Wing running INAV 7.1.2, i can observe a weird behavior of the horizon being significantly off.
It seems that there is an issue with the ahrs where the force from turns cannot be compensated

1. Artificial horizon is not level (see screenshot below)
2. When switching to angle mode, the aircraft is not level
3. When using a nav mode: turning to the opposite side than where the horizon drifted might not be possible. (When `nav_fw_bank_angle` is equal or close to the offset of the horizon, the max angle is in some cases already reached when flying straight.)

[image]

Default ahrs values are used:
```
set ahrs_dcm_kp = 2000
set ahrs_dcm_ki = 50
set ahrs_dcm_kp_mag = 2000
set ahrs_dcm_ki_mag = 50
set ahrs_acc_ignore_rate = 15
set ahrs_acc_ignore_slope = 5
set ahrs_gps_yaw_windcomp = ON
set ahrs_inertia_comp_method = ADAPTIVE
set ahrs_gps_yaw_weight = 100
```

## Steps to Reproduce

1. Fly the plane
2. Do a hard left turn
3. Revert to level flight
4. Observe artificial horizon being tilted to the right
5. Do a hard right turn
6. Revert to level flight
7. Observe artificial horizon being tilted to the left

Alternative:
1. Set the max bank angle to a relative low value
2. Start a waypoint mission with sharp turns
3. let the aircraft run for a while
4. Observe that the aircraft starts missing waypoints after a while as it cannot turn to one side anymore

## Expected behavior

The AHRS force compensation should counteract the g-forces occurring in hard turns. No horizon drift should occur.

## Additional context

CLI Dump: [Pastebin CLI Dump](https://pastebin.com/4y1EvDeZ)
Blackbox Log: [blackbox_log_2024-12-02_134803.TXT](https://github.com/user-attachments/files/17983474/blackbox_log_2024-12-02_134803.TXT)

### Board Name and Version:
```
# version
# INAV/TBS_LUCID_FC 7.1.2 Sep 23 2024 / 11:56:35 (8e7cb50c)
# GCC-10.3.1 20210824 (release)
```

Jetrell, 2024-12-04:
Vibrations from the motor/prop can cause accelerometer drift.. This often leads to `ahrs_acc_ignore_slope` providing a false report.
I noticed you have `gyro_lpf_hz = 60`.. There is always a risk of inducing drift if the filter low-pass cutoff is set too high, without your hardware being precisely balanced.
And as a side note. The effect of even accidentally mounting your flight controller on a vibration node point. Always has the possibility to make things worse when any of the filters low-passes are increased.

This doesn't mean there isn't a side issue that's making this more pronounced.. But until that gets resolved. Use the advise above.

b14ckyy, 2024-12-06:
Can you please check your config and look what value this setting is set to `ahrs_inertia_comp_method`
in CLI type: `get ahrs_inertia_comp_method` and tell us what it shows.

noahwaldner, 2024-12-08:
@b14ckyy 

it is set to adaptive (i think this is correct for wings?)

```
ahrs_inertia_comp_method = ADAPTIVE
```

there is a full dump in a pastebin linked in my initial issue report

b14ckyy, 2024-12-08:
yeah that is correct. 

the LOG shows nothing spectacular wrong. Very little noise. The only thing I could check is your complete flight. Can you upload the complete DVR of the entire flight please? Just to YT or so.

b14ckyy, 2024-12-18:
Any Update @noahwaldner?

b14ckyy, 2025-01-17:
closing as no feedback provided

closed 2025-01-17 by b14ckyy