iNavFlight/inav #10507: TBS Lucid H7 logging problems

rts18, 2024-12-06:
I brought this FC just a while ago. I've had little luck getting it to log data at greater than 3% without it causing the OSD to miss frames with my 3 profile settings.
Many times it won't log at all. And once when it did log at 50%, the ESC rebooted. 
Most of the logs have parsing errors. Not that I like posting logs online.

An interesting thing I did notice was if the `blackbox MOTORS` logging field was not chosen when using Dshot. It seem to operate better. But that ain't much good. I want to use Dshot and need to log the motor data or I can't tune the quad.

I tried a few SD cards, with the results being no different. I thought it might be related to this issue https://github.com/iNavFlight/inav/issues/7759. Which still causes my Matek H7 and F7 wing boards to experience the similar sort of problem.
@bkleiner Has there been any movement in this area?

- FC Board name and vendor: TBS_LUCID_H7
- INAV version string: 8.0.0 RC1 and RC2

bkleiner, 2024-12-06:
i have purchased some more sdcards to experiment with, but havent had time to really dig into it.
Please append your experience to the existing issue, not much point in having another one.

rts18, 2024-12-06:
>Please append your experience to the existing issue,

@bkleiner I'm not sure its related to a single issue.
One part seems to be initialization problems with some cards. And the other part seems to be Dshot and maybe the SDIO card access.
I'd almost say logging motor data is causing a bottle neck that is slowing other high demand tasks.

bkleiner, 2024-12-06:
that original ticket is also not related to a singular issue, from day one we were investigating high cpu load as well as out-right hangs during init.

so i don't see how your experience is significantly different from whats already reported there.

b14ckyy, 2024-12-06:
I have a H743-SLIM from matek waiting for me at the post office. So I can also join the tests with multiple SD cards that I use for logging

rts18, 2024-12-06:
This Lucid H7 has been a big disappointment to me because of this problem. Where I live, it wasn't a cheap flight controller.
I've spent days tracking down the data logging issue. 

Now I'm left wondering if I should replace it with an F405, so I can at least use the quad.
@bkleiner I figure you probably don't have time to fix the issue anytime soon? Since the other issue report is over 4 years old. 
I'll leave this report open for some new perspective.
I haven't yet reported it to TBS. They may not be aware of the problem.

b14ckyy, 2024-12-07:
I would like to have this topic more organized. So please continue (or maybe copy over your opening post) this here: https://github.com/iNavFlight/inav/issues/9918

as this thread was reported earlier we should keep it there to not miss information.

closed 2024-12-07 by b14ckyy

b14ckyy, 2024-12-08:
see https://github.com/iNavFlight/inav/issues/9918#issuecomment-2526152420 to test a potential fix.