iNavFlight/inav #10518: Rangefinder

Lalbabu2001, 2024-12-10:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [INAV Discord Server](https://discord.gg/peg2hhbYwN)
* [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
* [INAV Official on Telegram](https://t.me/INAVFlight)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

I attached vl53l0x rangefinder to have a good landing but the sensor is working as a normal barometer.
## Steps to Reproduce

1.just arm the quad and take of 
2.try to land in althold mode through throttle down 
3.
4.

## Expected behavior

As the rangefinder can detect ground with high accuracy, it can be used as soft landing tool,
What the idea is whenever the quad is in rangefinder coverage the quad use throttle command differently like max landing speed should be limited to 50 cm/sec (any value) even the throttle is pulled lowest this will avoid situations like jerky/bump landing. Throttle up command will work normally no changes there. And when the quad touches the ground it disarm the quad or lower the motor speed to idle.
## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

and-sh, 2024-12-11:
Your rangefinder didn't connected at all.
[image]

Look at debug mode AGL. It save following parameters

    DEBUG_SET(DEBUG_AGL, 0, posEstimator.surface.reliability * 1000);
    DEBUG_SET(DEBUG_AGL, 1, posEstimator.est.aglQual);
    DEBUG_SET(DEBUG_AGL, 2, posEstimator.est.aglAlt);
    DEBUG_SET(DEBUG_AGL, 3, posEstimator.est.aglVel);

and-sh, 2024-12-11:
It worth to look at debug Altitude too
DEBUG_SET(DEBUG_ALTITUDE, 0, posEstimator.est.pos.z);       // Position estimate
DEBUG_SET(DEBUG_ALTITUDE, 2, posEstimator.baro.alt);        // Baro altitude
DEBUG_SET(DEBUG_ALTITUDE, 4, posEstimator.gps.pos.z);       // GPS altitude
DEBUG_SET(DEBUG_ALTITUDE, 6, accGetVibrationLevel());       // Vibration level
DEBUG_SET(DEBUG_ALTITUDE, 1, posEstimator.est.vel.z);       // Vertical speed estimate
DEBUG_SET(DEBUG_ALTITUDE, 3, posEstimator.imu.accelNEU.z);  // Vertical acceleration on earth frame
DEBUG_SET(DEBUG_ALTITUDE, 5, posEstimator.gps.vel.z);       // GPS vertical speed
DEBUG_SET(DEBUG_ALTITUDE, 7, accGetClipCount());            // Clip count