iNavFlight/inav #10521: What abut use lockal Differential GPS?

and-sh, 2024-12-10:
## Current Behavior
A single gps receiver is used to determine position. 

## Desired Behavior
Not to costly to set second gps unit near operator position and transmit it`s readings to drone.
It can be easily done by such device without programming at all
https://micoair.com/radio_telemetry_lr24f/
Drone can correct onboard gps reading if stationary gps unit moves.
https://en.wikipedia.org/wiki/Differential_GPS

## Suggested Solution
Add dgps compatability.

## Who does this impact? Who is this for?
For people who want inav fly better than dji.

## Additional context
There is a suspicion that DJI uses a similar technology with the gps receiver in the smartphone. But for better results gps receivers must be identical.

sensei-hacker, 2024-12-10:
You can plug in a differential GPS just the same as any other. There's nothing for INAV to do here.
But be aware that differential GPS needs one of two conditions to be effective:

1. A base station / reference station at a location precisely known by optical/laser surveying
OR
2. A flight lasting several hours or more (especially flights lasting several months)

Note that condition #1 is NOT satisfied by checking the location of the base station via GPS. The entire point is to have accuracy better than GPS, so if you base your reference off the GPS on your phone or whatever you haven't gained anything - you still have regular GPS accuracy just with extra steps (and extra cost, and extra ways for it to break).
Condition #1 can be satisfied by hiring a surveying crew, or if there is a commercial service offering corrections with a surveyed reference station close to your flying location, you can subscribe to their feed.  The reference station does need to be nearby - the ionespheric delay in Dallas, Texas isn't relevant to a flight taking place in Florida.

About condition #2:
The ionespheric delay goes on a daily cycle and an annual cycle. The delay is greatest around noon and in November when the earth is closest to the Sun, lowest close to midnight and July, when the earth is far from the sun.
You said "Drone can correct onboard gps reading if stationary gps unit moves".  It will seem to move several centimeters if you take off at 9:00 AM and fly until noon, a three-hour flight.  It will move more if you take off in November and continue the flight until May or June.

So buying a differential GPS may be useful to you if your flights last several hours AND can't tolerate several centimeters of position error.  If your flights last minutes, there will be no significant change in the F layer of the ionospehere over the course of minutes. If you don't need better than 1-meter or 2-meter position accuracy, you wouldn't need differential GPS.

Another note on this.  Ionospeheric effects are primarily an issue when the satellites you're using are low to the horizon. Simply because the signal has to go through much more atmosphere if it's travelling laterally along the atmosphere, vs going directly through the shortest path perpendicular to the surface. When the US-sponsored GPS constellation was the only one available, some of the 15-16 visible satellites would be low to the horizon and be affected.  You want to use about 8-12 of them for navigation, so you may well end up with a couple low ones.
Now, modern modules such as the M9, and even the M8 or M10 can receive all four constellations. (with lower precision on the fourth constellation for M10, since it's single-receiver). You may well get 16 or more good satellites if you turn on two, three, or four constellations.  One or two that are low to the horizon aren't going to have the same effect as it did back in the days when GPS was the only constellation available.

In any event, there's nothing for INAV to do with this. Plug in a regular GPS module or with with differential - INAV doesn't know or care which you're using.

and-sh, 2024-12-11:
There are some observations. 
DGPS has been used in construction for a long time and it replaces the theodolite.
The DJI does not land at the take-off point even after 15 minutes of flight.

There aren't many flat spots in the field. It would be very convenient if the drone could return to where it took off from with an accuracy of at least 10 cm.

and-sh, 2024-12-13:
In any case, when I get the second GPS, I'll look at the correlations.
I thought that someone here had already done this.

and-sh, 2024-12-14:
https://m.youtube.com/watch?v=gceXRfFl4Sc
Someone already tried to do this

and-sh, 2024-12-14:
Ready to use solution
https://content.u-blox.com/sites/default/files/ZED-F9P_IntegrationManual_UBX-18010802.pdf
You need 2 modules and a modem to transmit corrections. The module has 2 ports, so you can integrate it without changing the software. Need bulky antennas.

sensei-hacker, 2024-12-14:
> Ready to use solution https://content.u-blox.com/sites/default/files/ZED-F9P_IntegrationManual_UBX-18010802.pdf You need 2 modules and a modem to transmit corrections. The module has 2 ports, so you can integrate it without changing the software. Need bulky antennas.

Yes exactly. And you need a source of appropriate RTCM data. That means either have a survey crew out to survey the location of your ground reference, or IF an NTRIP provider has a station near you, you can subscribe to their feed.

If your only source of information about the location of the reference unit is the GPS signals - well you could get that level of accuracy by skipping all the differential stuff. Just stick an M9 on your aircraft and you've good the same accuracy without any extra gear or extra complications.

and-sh, 2024-12-14:
In general, I was hoping for optic flow. But I couldn't get it to work and put it aside for now.

closed 2025-02-03 by DzikuVx