iNavFlight/inav #10527: Report M10 units not set to high performance mode

mmosca, 2024-12-12:
## Current Behavior
M10 GPS actually limits the update rate based on selected constellations and performance mode.

10Hz with all contellations is only possible in high performance mode.

[image]

High performance mode is configured in OTP (one time programmable memory) and requires a hardware reset before taking effect.

## Desired Behavior
Alert user and/or set update rate to a supported value.

## Suggested Solution
According to m10 [integration manual](https://content.u-blox.com/sites/default/files/MAX-M10S_IntegrationManual_UBX-20053088.pdf]), it is possible to verify if high performance mode is enabled by issues a VAL-GET message

From the integration manual "2.1.7 High performance navigation update rate configuration":
```
To verify that the configuration item is correctly set,
• send the following sequence to the receiver: B5 62 06 8B 14 00 00 04 00 00 01 00 A4 40 03
00 A4 40 05 00 A4 40 0A 00 A4 40 4C 15
• receive a UBX-CFG-VALGET message with the following sequence: B5 62 06 8B 24 00 01 04
00 00 01 00 A4 40 00 B0 71 0B 03 00 A4 40 00 B0 71 0B 05 00 A4 40 00 B0 71 0B 0A 00
A4 40 00 D8 B8 05 76 81
• receive a UBX-ACK-ACK message with the following sequence: B5 62 05 01 02 00 06 8B 99
C2
```

Add warning to status output about limited update rate in gps module.
Optionally, set update rate to match datasheet.

## Who does this impact? Who is this for?
Users of generic M10 gps 

## Additional context
M9 is a better gps unit than M10, but M10 is cheap an popular. I suspect users having problems with M10 gps in newer INAV versions may be affected by this.

sensei-hacker, 2024-12-12:
Good catch. The datasheet indicates the difference in power consumption between performance mode vs low power mode is 3-4ma @ 3V. So roughly 0.5 - 1 ma from the battery.

Meaning even on a two-hour flight, the difference would be 1-2 mah of power consumption. The M10 is a low power item even in the "performance" mode.

Perhaps we can figure that anyone choosing GPS auto config is going to want the higher performance and is willing to spend the 1 ma to get it. So just set that the same as we do other settings.

Of course people who have a highly unusual use case can choose not to auto config.

mmosca, 2024-12-12:
Given that I am not even sure you can set the configuration, as it is stored on OTP, we should at least flag units that are not in high performance mode so users know they should either stick to GPS + Galileo or lower the update rate to match whatever they selected

sensei-hacker, 2024-12-13:
Ah yes if we can't update it, that makes to me also.
My M10 test unit is broken at the moment. :(