iNavFlight/inav #10529: NEW TARGET: FOXEER F405V2

66nola66, 2024-12-12:
Propose addition of new target asset FOXEER F405V2

MrD-RC, 2024-12-12:
Are you a representative of Foxeer?

mmosca, 2024-12-12:
Also, have a look at the new hardware policy: https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

66nola66, 2024-12-12:
No i just bought one and realized its not supported, i like inav. But better ardu.....so hope to see it soon

Yahoo Mail: Search, Organize, Conquer

MrD-RC, 2024-12-12:
> No i just bought one and realized its not supported, i like inav. But better ardu.....so hope to see it soon

You should contact Foxeer and let them know that you would like an INAV target for this flight controller.

66nola66, 2024-12-13:
Nope, flying good on ardu.....no reason

66nola66, 2024-12-13:
Yeah, it works just fine on ardu, im gonna remove tjis request

66nola66, 2024-12-13:
Not needed

closed 2024-12-13 by 66nola66