iNavFlight/inav #10530: Feature Request: Rotate camera/OSD orientation

bdussy, 2024-12-13:
## Current Behavior
I can't find a way to rotate the osd/camera. I have the camera turned 90 degrees than how it usually would be.

## Desired Behavior
I would like to rotate the camera direction in the settings somewhere.

## Suggested Solution
Add a option in inav to rotate camera 

## Who does this impact? Who is this for?
I'm using this shitty camera mount. The hole for the wire is on the side instead of the top. So the camera is mounted at a 90 degrees angle. 
[image]
[image]
(https://www.aliexpress.us/item/3256807041085568.html)

## Additional context

bdussy, 2024-12-13:
It is analog. CADDXFPV Ratel Pro. https://manuals.plus/m/3de21995d5acbb32bf855cba02094eab19ada0757f49135524e5c7a85c6aa49f_optim.pdf
I could flip the camera horizontal and vertical but the doesn't help the issue. 
I can turn the screen/monitor 90 degrees. I would be happy if the OSD lines/fonts can be rotated. so I'm not reading the numbers sideways. but no worries if it is a difficult thing to do. I might just live with it or cut the mount and stick the camera on less securely
thanks for your help

mmosca, 2024-12-13:
The chip that does osd has a fixed font orientation and grid size, so that is a major deal breaker. Maybe you can find a way to mount the camera in the correct orientation.

closed 2024-12-13 by mmosca

bdussy, 2024-12-14:
I appreciate your insight