iNavFlight/inav #10532: Inconsistent Throttle and Disarm Behavior with MSP_SET_RAW_RC

netIOTil, 2024-12-13:
Hey, 

I’m experiencing unexpected behavior when controlling my flight controller via a Linux device connected through USB-C. I don’t use an external receiver; instead, my Linux device sends MSP commands directly to the flight controller.

## Current Behavior

Using MSP_SET_RAW_RC, I send the following channel values:
Disarmed: 
channels = [1500, 1500, 1000, 1500, AUX1_OFF (1000), 1000, 1000, 1000, 1000, 1000, 1000, 1000]

Armed:
channels = [1500, 1500, 1000, 1500, AUX1_ON (1830), 1000, 1000, 1000, 1000, 1000, 1000, 1000]

Once armed, increasing the throttle works fine. For example:

1. Setting throttle to 1500 works as expected.
2. Reducing throttle back to 1100 also works fine.

However, after I increase the throttle to 1500, any command to set the throttle below 1100 (including disarm commands) does not work. Specifically:
- Sending a disarm command (e.g., AUX1_OFF = 1000) has no effect.
- Setting the throttle to 1000 has no effect.

but I still can increase the throttle level to any value.

## Steps to Reproduce

1. Send MSP_SET_RAW_RC with disarm values:
channels = [1500, 1500, 1000, 1500, AUX1_OFF (1000), 1000, 1000, 1000, 1000, 1000, 1000, 1000]

2. Send MSP_SET_RAW_RC with arm values:
channels = [1500, 1500, 1000, 1500, AUX1_ON (1830), 1000, 1000, 1000, 1000, 1000, 1000, 1000]

3. Send MSP_SET_RAW_RC with throttle set to 1500.
4. Send MSP_SET_RAW_RC with throttle set to 1100.
5. Send MSP_SET_RAW_RC with throttle set to 1000 or disarm values:
channels = [1500, 1500, 1000, 1500, AUX1_OFF (1000), 1000, 1000, 1000, 1000, 1000, 1000, 1000]

## Expected behavior

- The motors should decrease speed when the throttle is set below 1100.
- Sending a disarm command should disarm the flight controller and stop the motors.

- FC Board name and vendor:  speedybee F405 V3 stack with 60A 4-in-1 ESC
- INAV version string: 7.1.2

stronnag, 2024-12-13:
Disarming via switch / `MSP_SET_RAW_RC` works just fine the SITL and [msp_set_rx](https://github.com/stronnag/msp_set_rx)

In order for us to take this issue more seriously, you should:
(a) Provide a link to the your code  causes this issue
(b) Provide a CLI diff for the vehicle on which the problem occurs.

b14ckyy, 2024-12-13:
What stronnag said. 
I would even consider the DIFF as the most important part here as a first point of failure analysis. There are a few settings affecting the exact behavior you are describing: 

disarm_always - Defaults to ON. If set to OFF it will not allow disarming if throttle is not below min_check.
feature MOTOR_STOP - is by firmware default and multirotor presets disabled. This will prevent the motors from stopping if throttle is below min_check 

if both options are disabled, you need the throttle channel at 1000 and the arming channel in disarmed state both together to disarm. 

there are a few more settings that might affect the behavior so a DIFF is important. Maybe better even a DUMP.

netIOTil, 2024-12-14:
[dump.txt](https://github.com/user-attachments/files/18135386/dump.txt)

hey b14ckyy, thanks for your resopnse!!

here the DIFF, and I also add the dump as dump.txt file.

also just the add up, is not just disarm that stop responding even lowering the throttle bellow 1100 is not responding.

############################################################
Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# INAV/SPEEDYBEEF405V3 7.1.2 Jun  1 2024 / 00:45:26 (4e1e59eb)
# GCC-10.3.1 20210824 (release)

# start the command batch
batch start

# resources

# Timer overrides

# Outputs [servo]

# safehome

# Fixed Wing Approach

# features
feature PWM_OUTPUT_ENABLE

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y

# Receiver: Channel map

# Ports
serial 1 0 115200 115200 0 115200
serial 3 0 115200 115200 0 115200

# LEDs

# LED color

# LED mode_color

# Modes [aux]
aux 0 0 0 1825 2100

# Adjustments [adjrange]

# Receiver rxrange

# temp_sensor

# Mission Control Waypoints [wp]
#wp 0 invalid

# OSD [osd_layout]

# Programming: logic

# Programming: global variables

# Programming: PID controllers

# OSD: custom elements

# master
set looptime = 500
set gyro_main_lpf_hz = 90
set gyro_main_lpf_type = PT1
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 70
set setpoint_kalman_q = 200
set gyro_zero_z = -1
set acc_hardware = BMI270
set align_mag = CW270FLIP
set mag_hardware = NONE
set baro_hardware = SPL06
set receiver_type = MSP
set blackbox_rate_denom = 2
set motor_pwm_protocol = DSHOT300
set failsafe_delay = 200
set failsafe_procedure = DROP
set applied_defaults = 6
set airmode_type = THROTTLE_THRESHOLD

# mixer_profile
mixer_profile 1

set model_preview_type = 3

# Mixer: motor mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# Mixer: servo mixer

# profile
profile 1

set mc_p_pitch = 32
set mc_i_pitch = 90
set mc_d_pitch = 25
set mc_p_roll = 28
set mc_i_roll = 80
set mc_p_yaw = 30
set mc_i_yaw = 80
set dterm_lpf_hz = 85
set dterm_lpf_type = PT3
set d_boost_min =  1.000
set d_boost_max =  1.000
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set smith_predictor_delay =  1.500
set tpa_rate = 20
set tpa_breakpoint = 1200
set rc_expo = 75
set rc_yaw_expo = 75
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60
set ez_filter_hz = 90

# battery_profile
battery_profile 1

set throttle_idle =  2.000

# save configuration
save

# end the command batch
batch end

b14ckyy, 2024-12-14:
AS I expected: 
- `MOTOR_STOP` is disabled, this keeps the motors spinning if throttle channel is at 1000. They will spin at `throttle_idle` percent + `min_command` and depends if your ESCs are calibrated (if no DSHOT used)
- If you raise the throttle above a certain value (can't remember if that is configurable) the PID Controller Integral will kick in and the PID stabilization stays active. Lowering the throttle on the bench with no props on will still keep I-Term ramping up as the airframe is not following the setpoint when not flying. This is normal. 
- Disarming should work though because `disarm_kill_switch = ON` not sure what the problem is there. What mode are you in when you test that?

netIOTil, 2024-12-14:
Now I see that disarm does work when I only(no other extra messgae) send the disarm command at a 0.02-second interval (time.sleep(0.02).

Can you please explain if there is a message queue?
What is the correct message rate I should use?

Currently, I only send a msp command after receiving a response(some type of message/ response loop), but I am observing behavior that I don’t fully understand. 
Because I didn’t (and still don’t fully) understand the behavior of MSP_SET_RAW_RC, I started developing my own control logic, stabilization algorithms, and motor mixer on the Linux(that is connected via USB C to the RC) side and now I try only to use MSP_SET_MOTOR to control the motors.

However, I noticed the following behavior:

When I send an ARM command using MSP_SET_RAW_RC with the payload
channels = [1500, 1500, 1000, 1500, AUX1_ON, 1000, 1000, 1000, 1000, 1000, 1000, 1000]

at a 0.02-second interval (time.sleep(0.02), and then switch to MSP_SET_MOTOR, it takes some time for MSP_SET_MOTOR commands to take effect.
Once MSP_SET_MOTOR commands start working, I can change the motor values every 0.02 seconds without delay.

Could you please explain why this happens?