iNavFlight/inav #10534: battery sensor not working on inav 8 rc3, matek f405 wte

sys512, 2024-12-15:
## Current Behavior
Battery voltage shown as constant 0 even when battery is connected

## Steps to Reproduce

1.Install INAV 8.0.0 RC3 on Matek F405 WTE
2.Configure battery
3.Look at battery voltage
4.Look at the battery voltage again

## Expected behavior
Correct battery voltage measurement shown.

---

- FC Board name and vendor: Matek F405 WTE
- INAV version string: INAV/MATEKF405TE_SD 8.0.0 RC3

MrD-RC, 2024-12-15:
Is this in configurator or on the OSD?

sys512, 2024-12-15:
configurator

sys512, 2024-12-15:
i should post this issue in inav configurator repo?

MrD-RC, 2024-12-15:
I can move it. Do you see the voltage on the OSD?

sys512, 2024-12-16:
I already rolled back to 7.1.2 but ok I'll update and check wait

Itay-Sharoni, 2025-01-09:
Can you make sure UART6 is on MSP mode? i had similar issue but it was due to UART6 was not on MSP mode
Im using SpeedyBee f405 Wing, which is quite similar i believe

(or maybe it is different UART on this FC, just compare the ports sections between 7.12 and 8.0.0)

worth checking :)