iNavFlight/inav #10537: 1080 HD OSD -MSP left side of stats labels missing

pitts-mo, 2024-12-15:
## Current Behavior
left side of OSD stats labels are missing when the current OSD profile contains only the system messages element centered on screen. I suspect this issue will occur when there are not any OSD elements on the left side of the current OSD profile.
[image]

[image]

## Steps to Reproduce

1.
2.
3.
4Note my build was with 1080

## Expected behavior
The expected behavior is to see the entire OSD stats panel when OSD is enabled regardless of OSD element placements within the current OSD profile.

Temporary Workaround:
The system messages element dragged to the left edge of the OSD profile. 
[image]

[image]

## Suggested solution(s)
Not sure exactly where this is being triggered.

## Additional context

---

- FC Board name and vendor:
- INAV version string:

MrD-RC, 2024-12-15:
To me it looks like the stats screen is being clipped to 4:3. There is also text missing from the right side of the stats. 

The system messages is drawn by something completely different to the stats screen. I don't get why this clipping is occurring when the system messages are moved. But maybe having it left switches to 16:9, but centred it drops to 4:3?

I'll see what I can check out in HITL and with Walksnail.

pitts-mo, 2024-12-16:
This issue has been resolved in PR #10540

aviphysics, 2024-12-18:
The low res OSD was added to improve OSD refresh rate, especially for AHI. This is a bandwidth limitation and I believe affects all HDZero devices. I expect there will be a number of upset people if the option is removed. People that use AHI get really annoyed when the refresh rate is low.

I think @geoffsim is the expert on this code and should be consulted before making changes.

In about a week I will have some time to test the reported behavior.

@pitts-mo it would be helpful to know if you are using goggles or VRX.

pitts-mo, 2024-12-18:
I use artificial horizon with HDZero Goggles.

When I glanced at the code I suspected the need for this feature was already resolved now that INAV receives the display port size from the VTX.

If a similar feature needs to be re-added I think we will need to set it static or at least enhance the logic as to prevent it from erroneously clipping the sides of the OSD.

aviphysics, 2024-12-18:
@pitts-mo you need to check what the VTX code is doing.

My recollection is that the performance improvement was from the reduced bandwidth between VTX and VRX

pitts-mo, 2024-12-19:
> @pitts-mo you need to check what the VTX code is doing. 
? Sorry, I typically only have small and sporadic bits of hobby time. 

Seems like a comparison of update rate between an OSD profile with only artificial horizon centered and another OSD profile with only artificial horizon left might be helpful. (I might be able to look into this on Sunday)

I see @mmosca's comment on PR #10540 , INAV does not yet support getting display size from VTX and his suggestion of an SD mode makes a lot of sense if this is still needed.

pitts-mo, 2024-12-19:
issue #8006 and PR #8264 appear to be the most recent items related to this.

pitts-mo, 2024-12-20:
Tested inav 8.0.0-rc3 with my normal OSD profile, OSD profile w/AHI only, and OSD profile w/AHI & pilot name  upper left. I am unable to perceive any AHI performance difference between these three profiles. In fast movement the outer dots of the AHI were so fast they were almost impossible to see.

I am using HDZero Goggle and Race v3 VTX running recent builds from master.

If someone reports they are still affected by the old slow OSD problem we can navigate toward a more appropriate implementation or solution.

aviphysics, 2024-12-20:
Did you test in 1080p 30 only, or also 720p 60?

pitts-mo, 2024-12-21:
I had performed the actual comparison at 1080p but I did also verify the AHI updated extremely fast in 720p mode as well. 

The #8006 issue suggests to me that HDZeros OSD update rate at the time was horrible regardless of FC firmware. The timing of code changes makes me suspect HDZero users running pre v1.0.0 firmware would experience the old lag issue.

@aviphysics If your concern persists please provide the test conditions where you find the slow OSD rate still exists.

closed 2025-01-17 by b14ckyy