iNavFlight/inav #10543: FC MATEK 405 wing V2 - INav 7.1.2 - lost telemetry sensors

Asafsol, 2024-12-17:
## Current Behavior
Im using frsky X18 and tdmx rx, i used to have all the gps sensors on the tx
in the process of trying to make the "sats" opion to work i tried to move the telemetry s.port to UART6 and then back to UART 2 - but then tried to del sensors and rediscover and they gone.
So far nothing helped to regain the gps sensors

any ideas?

MrD-RC, 2024-12-17:
There isn't a `sats` sensor with SmartPort telemetry. There is `GPS` which gives the coordinates. There is also a sensor currently called `Tmp1` which gives various information about the GPS link. Including the number of sats, connection quality, and fix status. This sensor is renamed `480` in INAV 8.0. To stop ETHOS from converting it to a temperature sensor. https://github.com/iNavFlight/inav/blob/master/docs/Telemetry.md#available-smartport-sport-sensors

There is a widget that you can use. This extracts the number of sats from the `480` or `Tmp1` telemetry sensors. https://github.com/MrD-RC/ETHOS-GNSS-display-for-INAV

The `sats` telemetry sensor is most commonly found in the CRSF telemetry. 

The sensors missing after switching UARTs can only be a setup issues. As it was working fine before changing it.

b14ckyy, 2024-12-18:
Please provide a diff from INAV, the exact type of Receiver you use and how its connected and what protocol does it use (Sbus+S.Port, FPORT, FPORT2, FBUS)

Asafsol, 2024-12-18:
Using frsky TDMX rx
Using Sbus + S.Port
Add diff list
[Uploading INAV_7.1.2_cli_20241218_211944.txt…]()

b14ckyy, 2024-12-18:
Your Ports are probably configured incorrectly.

First of all its a F405 MCU so no inverter except on the dedicated sbus pad. The TDMX also has no inverted pads. That means you can ONLY run SmartPort Telemetry on a softserial that can be inverted in software. 

you need to enable Softserial on the FC, then enable SmartPort on the Softserial port in the Ports Tab and connect the SmartPort wire to Tx2 on the FC. Don't be confused, when Softserial is enabled, Tx2 becomes Softserial1. 
You keep your sbus wire on the SBUS pad on the FC and Serial Receiver on UART2

Asafsol, 2024-12-18:
This is how everything is wired
Ive enabled softserial and smartport - not change in the telemetry

b14ckyy, 2024-12-18:
that does not match your DIFF. 
can you show me a screenshot of your ports page please?

Asafsol, 2024-12-18:
Cause ive just changed it now
[image]

b14ckyy, 2024-12-18:
Why do you have MSP on soft serial? If you do that, when disarmed that UART will only communicate in MSP and telemetry only switches on when armed. Disable MSP on SS1, that makes no sense.

Asafsol, 2024-12-18:
ok i removed msp from uart 1
still no change in telemetry

b14ckyy, 2024-12-18:
Have you tried to rediscover sensors? 
Did you check your receiver settings from the radio that Telemetry is actually on?

Asafsol, 2024-12-19:
Yea i did

b14ckyy, 2024-12-19:
Then I am out of ideas. Something must be at fault in your setup. Maybe a broken wire or so, no idea. Maybe doublecheck everything.

Asafsol, 2024-12-20:
Just saw that the fc dosent show batt volt. Its shows zero

any idea why?
Maybe its something to do with the telemetry?