iNavFlight/inav #10547: Some misunderstanding in calculation climb rate

and-sh, 2024-12-21:
I don't really understand why there is an iddle here, because this will lead to the maximum speed down being greater than the speed up?
https://github.com/iNavFlight/inav/blob/8.0.0-RC3/src/main/navigation/navigation_multicopter.c#L153
[image]

MrD-RC, 2024-12-21:
Looks to me like it's scalling here.

and-sh, 2024-12-21:
It seems to me that the idle is added in the mixer and not in the receiver and it has nothing to do with the position of the sticks

and-sh, 2024-12-21:
This is what it looks like. The descent speed is much greater than ascend.
[image]

and-sh, 2024-12-26:
By the way, the Max. Alt-hold climb rate is set to 150

and-sh, 2024-12-31:
This does not work. Descend speed to low. 
[image]
Now I guess that rcControlsConfig()->alt_hold_deadband here is unnecessary, because it was applied above at line 140
[image]

and-sh, 2025-01-06:
Fine. Now it works as I like. Manual climb rate set to 100.
[image]
[image]
I don't know where the 15 percent of the descent speed went. At least it's not a several times.
Changes
[image]

and-sh, 2025-01-08:
Probably no one needs it

closed 2025-01-08 by and-sh