iNavFlight/inav #10551: DS18B20 Temp sensors still not detected in 8.0

dcan999, 2024-12-22:
I made a report earlier this year about the DS18B20 temp sensors not working with 7.x and never got a response from any Devs about it, just one user who said they had to define the temp sensors in their own custom target to make it work. I was hoping this would be fixed in 8.x but it appears they still do not work. Can this be fixed?

Thank you

MrD-RC, 2024-12-24:
Where was that issue raised?
https://github.com/iNavFlight/inav/issues?q=+is%3Aissue+author%3Adcan999

dcan999, 2024-12-24:
I first brought it up here as a discussion since I didn't know if it was an actual issue or just user error, I guess should have made it an official issue after another user confirmed it was, I figured Devs monitor the discussions. 

https://github.com/iNavFlight/inav/discussions/9732

I also just commented on this thread as I took it as native DS18B20 support (without a DS2482 breakout needed) was implemented, as the thread was closed as complete, but that doesn't seem to have happened either. 

https://github.com/iNavFlight/inav/issues/4571