iNavFlight/inav #10553: Unclear about RM3100 magnetometer & Benawake TFMiniPlus both running under I2C port on iNav

cramredan1, 2024-12-23:
I'm short on available UARTS and can't get my RM3100 compass +Benawake TFMINI-PLUS rangefinder modules to be detected by iNav 8.0 over I2C. Current documentation does not provide any clue on that topic. 

**Are these modules taken in charge by iNav through the I2C communication protocol?**

Many thanks for your clear and precise answer.

sensei-hacker, 2024-12-23:
The RM3100 is supported, from what I can tell.  There are two Matek boards which define that as being attached via SPI.
Which flight controller are you using?

The TF Mini and Mini Plus are supported as well (same data protocol), on uart only as far as I can see.

cramredan1, 2024-12-24:
Thanks Sensei for your kind feedback. 
> RM3100... what do you mean by "what i can tell"? Did you try it yourself over I2C? Did you read it in official documentation? Where?
> TFMINIPLUS... you confirm what I suspected, UART connection only. May I ask you where/how to check I2C is not supported on this device (apart from wiring it) ?

sensei-hacker, 2024-12-24:
I looked at the code.
I looked briefly, so I could have missed something.

closed 2025-01-03 by sensei-hacker