iNavFlight/inav #10554: Random GPS failure message on boot

jaroszmm, 2024-12-23:
Sometimes, when I power up my plane with INAV 7 or 8 RC2 (tried on both) I get long constant beeping and GPS Failure message on OSD.
Sometimes GPS will connect after a short while, sometimes after a longer time. Sometimes it needs rebooting. Today however it wouldn't connect at all at the field and bench on 8 RC2.

I had to go into configurator, in PORTS disable GPS as sensor, reboot, go again into PORTS, enable GPS as sensor again, reboot - GPS works fine.

GPS is a solid BE250 module I've had for quite some time now. I also notice this on some other builds that sometimes INAV takes a longer while to actually communicate and realise that there is a GPS connected to the FC.

FC is Matek F405WMN on either 7.1.2 Stable or 8 RC2

Jetrell, 2024-12-28:
I've also had this happen. And I can't pin down any consistent cause. I see it while using various GNSS modules.. Some of the more common ones are - Matek M8Q's, Walksnail M181.
Admittedly, I do spend stacks of time flying and testing. So this maybe why I see it more often.

At maybe 1 time in 20 the modules will get a stack of com errors.

Then other times, I can come down from a flight with 22 Sat's.. Unplug the power to the FC.. Then when I power it up again within 5 minutes. It takes many minutes to get a fix again.. While normally, it gets that number of Sats again within a few seconds.

This isn't just with one module, which could be excused for having a bad backup battery.. It happens with many modules of different types. And at random intervals.
When this does occur. The so called hot start reboot, takes many times longer to get a fix, then the first cold start boot did.
I seen t occur with both analog and digital video systems.

Jetrell, 2024-12-28:
I've also had this happen. And I can't pin down any consistent cause. I see it while using various GNSS modules.. Some of the more common ones are - Matek M8Q's, Walksnail M181.
Admittedly, I do spend stacks of time flying and testing. So this maybe why I see it more often.

At maybe 1 time in 20 the modules will get a stack of com errors.

Then other times, I can come down from a flight with 22 Sat's.. Unplug the power to the FC.. Then when I power it up again within 5 minutes. It takes many minutes to get a fix again.. While normally, it gets that number of Sats again within a few seconds.

This isn't just with one module, which could be excused for having a bad backup battery.. It happens with many modules of different types. And at random intervals.
When this does occur. The so called hot start reboot, takes many times longer to get a fix, then the first cold start boot did.
I've seen it occur with both analog and digital video systems.